WebODM/WebODM #912: Time remaining bar required

manand881, 2020-09-19:
On many occasions while processing maps I've noticed that there is a bar that shows me a pictorial representation of how much of the task is completed but there aren't any stats of how much time is left for the task to finish. This feature is extremely important while processing large maps.

pierotofy, 2020-09-19:
Hi @manand881,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-09-22 06:20) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

smathermather, 2020-09-19:
Any time bar is a lie, and doubly so if we did one for OpenDroneMap :D. Prediction of time here is an impossible task.

That said, we might be able to get closer to this with lots of tests across lots of machines. odm-benchmarks is such a sub-project: https://github.com/OpenDroneMap/odm-benchmarks/ -- a catalog of tests on different machines with different test datasets.

Please contribute! Also, for sub-tasks in processing, I have started a very nacent document on sub-task processing time here: https://github.com/OpenDroneMap/odm-benchmarks/issues/10. My work here will ultimately focus on massive split-merge projects.

manand881, 2020-09-26:
Thank you so much for your time. I've started processing the sample datasets on the benchmark git, and I've open 1 pull request there which is the benchmark generated on my machine. will be bench marking more datasets as and when its possible

smathermather, 2020-09-26:
Awesome. Super useful and thank you.

pierotofy, 2022-10-19:
I'm closing this in favor of accepting a pull request for the above improvements.

closed 2022-10-19 by pierotofy

smathermather, 2022-10-20:
Agreed. And regardless, the first step is to have reasonable estimates of percentages so we can project time, and we don't have that yet.