WebODM/WebODM #913: Can't upload image to webodm hosted on a remote Ubuntu server

Rexsunon, 2020-09-20:
Hey guy. I uploaded webodm to a remote server on the cloud and when ever I try creating a task, uploading an image it excrees the max number of retries and throws an error in the console.

But I tried this locally before pushing to the server it worked perfectly. Is there a problem with running webodm on a remote server? Cause I've tried every fix I could find on the community and other github issues. I even deleted the server and rebuild it, re-upload webodm and all. But same issue

pierotofy, 2020-09-20:
Hi @Rexsunon,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-09-23 12:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Rexsunon, 2020-09-20:
**What's your browser and operating system**?
Google Chrome, Windows 10 64bit

**What is the problem**?
Hey guy. I uploaded webodm to a remote server on the cloud and whenever I try creating a task, uploading an image it exceeds the max number of retries and throws an error in the console.

But I tried this locally before pushing to the server it worked perfectly. Is there a problem with running webodm on a remote server? Cause I've tried every fix I could find on the community and other GitHub issues. I even deleted the server and rebuild it, re-upload webodm, and all. But the same issue

_**Log**_
```worker      | INFO Canceling Task [Task of 2020-09-20T15:55:12.306Z] (ebac1776-32b1-4a3f-b2a9-21c4414f9bab)
worker      | [2020-09-20 16:28:14,141: INFO/ForkPoolWorker-1] worker.tasks.process_task[079448cc-47bb-45e7-9e74-74590915f809]: Canceling Task [Task of 2020-09-20T15:55:12.306Z] (ebac1776-32b1-4a3f-b2a9-21c4414f9bab)
worker      | INFO Removing Task [Task of 2020-09-20T15:55:12.306Z] (ebac1776-32b1-4a3f-b2a9-21c4414f9bab)
worker      | [2020-09-20 16:28:18,337: INFO/ForkPoolWorker-1] worker.tasks.process_task[91434748-2ec2-4c68-b4ed-57910123441d]: Removing Task [Task of 2020-09-20T15:55:12.306Z] (ebac1776-32b1-4a3f-b2a9-21c4414f9bab)
worker      | WARNING [Errno 2] No such file or directory: '/webodm/app/media/project/15/task/ebac1776-32b1-4a3f-b2a9-21c4414f9bab/'
worker      | [2020-09-20 16:28:18,341: WARNING/ForkPoolWorker-1] worker.tasks.process_task[91434748-2ec2-4c68-b4ed-57910123441d]: [Errno 2] No such file or directory: '/webodm/app/media/project/15/task/ebac1776-32b1-4a3f-b2a9-21c4414f9bab/'
webapp      | WARNING Not Found: /api/projects/15/tasks/ebac1776-32b1-4a3f-b2a9-21c4414f9bab/```

**What should be the expected behavior**?
I expected the images to upload successfully and start processing

**Google Drive Images**
https://drive.google.com/file/d/18uvz35Vb1f1K0hHMfqhH29KIop-y7UDU/view?usp=sharing (error log)
https://drive.google.com/file/d/1ivRs9mzgzd8DeMPGgWLqMGkEjMHM9J4_/view?usp=sharing (browser error)

pierotofy, 2020-09-23:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: It looks like it might be a configuration / machine setup problem, not a bug per-se.

Make sure to include details of the machine when you open a topic there (machine specs, free disk space, RAM, etc.). :+1:

closed 2020-09-23 by pierotofy