WebODM/WebODM #915: imp module is deprecated in favour of importlib

BlueRoar, 2020-09-22:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Edge 85 on Windows 10

### What is the problem?
Ran my first batch,  and got 

`Process exited with code 1` error message

And the below in the Task Output

`/usr/local/lib/python3.6/dist-packages/sklearn/externals/joblib/externals/cloudpickle/cloudpickle.py:47: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses`

### What should be the expected behavior?

It should make a 3D model :) 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

run a dataset, no resize, options: 3d model

pierotofy, 2020-09-23:
Hey @BlueRoar :hand: could you copy/paste the entire task output?

BlueRoar, 2020-09-24:
```
... output truncated at 500 lines ...
[INFO]    Filtering /var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_139.ply (statistical, meanK 16, standard deviation 2.5)
[INFO]    running pdal translate -i "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_139.ply" -o "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_139_filtered.ply" outlier range --writers.ply.sized_types=false --writers.ply.storage_mode='little endian' --writers.ply.dims="x=float,y=float,z=float,nx=float,ny=float,nz=float,red=uchar,blue=uchar,green=uchar" --filters.outlier.method='statistical' --filters.outlier.mean_k=16 --filters.outlier.multiplier=2.5 --filters.range.limits='Classification![7:7]'
[INFO]    Filtering /var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_324.ply (statistical, meanK 16, standard deviation 2.5)
[INFO]    running pdal translate -i "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_324.ply" -o "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_324_filtered.ply" outlier range --writers.ply.sized_types=false --writers.ply.storage_mode='little endian' --writers.ply.dims="x=float,y=float,z=float,nx=float,ny=float,nz=float,red=uchar,blue=uchar,green=uchar" --filters.outlier.method='statistical' --filters.outlier.mean_k=16 --filters.outlier.multiplier=2.5 --filters.range.limits='Classification![7:7]'
[INFO]    Filtering /var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_263.ply (statistical, meanK 16, standard deviation 2.5)
[INFO]    running pdal translate -i "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_263.ply" -o "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_263_filtered.ply" outlier range --writers.ply.sized_types=false --writers.ply.storage_mode='little endian' --writers.ply.dims="x=float,y=float,z=float,nx=float,ny=float,nz=float,red=uchar,blue=uchar,green=uchar" --filters.outlier.method='statistical' --filters.outlier.mean_k=16 --filters.outlier.multiplier=2.5 --filters.range.limits='Classification![7:7]'
[INFO]    Filtering /var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_72.ply (statistical, meanK 16, standard deviation 2.5)
[INFO]    running pdal translate -i "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_72.ply" -o "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_72_filtered.ply" outlier range --writers.ply.sized_types=false --writers.ply.storage_mode='little endian' --writers.ply.dims="x=float,y=float,z=float,nx=float,ny=float,nz=float,red=uchar,blue=uchar,green=uchar" --filters.outlier.method='statistical' --filters.outlier.mean_k=16 --filters.outlier.multiplier=2.5 --filters.range.limits='Classification![7:7]'
[INFO]    Filtering /var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_53.ply (statistical, meanK 16, standard deviation 2.5)
[INFO]    running pdal translate -i "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_53.ply" -o "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_53_filtered.ply" outlier range --writers.ply.sized_types=false --writers.ply.storage_mode='little endian' --writers.ply.dims="x=float,y=float,z=float,nx=float,ny=float,nz=float,red=uchar,blue=uchar,green=uchar" --filters.outlier.method='statistical' --filters.outlier.mean_k=16 --filters.outlier.multiplier=2.5 --filters.range.limits='Classification![7:7]'
[INFO]    Filtering /var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_113.ply (statistical, meanK 16, standard deviation 2.5)
[INFO]    running pdal translate -i "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_113.ply" -o "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_113_filtered.ply" outlier range --writers.ply.sized_types=false --writers.ply.storage_mode='little endian' --writers.ply.dims="x=float,y=float,z=float,nx=float,ny=float,nz=float,red=uchar,blue=uchar,green=uchar" --filters.outlier.method='statistical' --filters.outlier.mean_k=16 --filters.outlier.multiplier=2.5 --filters.range.limits='Classification![7:7]'
[INFO]    Filtering /var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_378.ply (statistical, meanK 16, standard deviation 2.5)
[INFO]    running pdal translate -i "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_378.ply" -o "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_378_filtered.ply" outlier range --writers.ply.sized_types=false --writers.ply.storage_mode='little endian' --writers.ply.dims="x=float,y=float,z=float,nx=float,ny=float,nz=float,red=uchar,blue=uchar,green=uchar" --filters.outlier.method='statistical' --filters.outlier.mean_k=16 --filters.outlier.multiplier=2.5 --filters.range.limits='Classification![7:7]'
[INFO]    Filtering /var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_375.ply (statistical, meanK 16, standard deviation 2.5)
[INFO]    running pdal translate -i "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_375.ply" -o "/var/www/data/750d6eb1-ed94-487d-a951-42eb0efa0b7a/odm_filterpoints/parts/part_375_filtered.ply" outlier range --writers.ply.sized_ty [... cut ...]
File "/code/opendm/types.py", line 337, in run
self.process(self.args, outputs)
File "/code/stages/mvstex.py", line 115, in process
'-n {nadirWeight}'.format(**kwargs))
File "/code/opendm/system.py", line 79, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
/usr/local/lib/python3.6/dist-packages/sklearn/externals/joblib/externals/cloudpickle/cloudpickle.py:47: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
import imp
```

pierotofy, 2020-09-24:
`Exception: Child returned 137` means you ran out of memory. You need to allocate more memory to docker to process this dataset.

Since this is not a bug, I'm going to close this. :pray:

closed 2020-09-24 by pierotofy