WebODM/WebODM #917: Unable to execute GRASS script errors on plugins in fresh installation of WebODM running in Docker on MacOS

taue2512, 2020-09-29:
After having installed WebODM using the preconfigured WebODM Manager 1.2 on MacOS 10.15.7 inside a Docker environment, I am facing the following issues and error messages while using the two plugins - the rest of the installation works flawlessly:

Plugin Elevation Map (elevationmap 1.0.0):
`Could not execute GRASS script /webodm/plugins/elevationmap/elevationmap.py from /webodm/app/media/tmp/tmpb2jpym48_grass_engine: Starting GRASS GIS... Creating new GRASS GIS location <location>... Cleaning up temporary files... Executing <python /webodm/plugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/2/task/8d15a99a-63e4-48e4-a564-d3c9de27551b/assets/odm_dem/dsm.tif interval=1 format=GeoJSON noise_filter_size=0.0 epsg=4326> ... Traceback (most recent call last): File "/webodm/plugins/elevationmap/elevationmap.py", line 48, in <module> import cv2, math, argparse ModuleNotFoundError: No module named 'cv2' Execution of <python /webodm/plugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/2/task/8d15a99a-63e4-48e4-a564-d3c9de27551b/assets/odm_dem/dsm.tif interval=1 format=GeoJSON noise_filter_size=0.0 epsg=4326> finished. Cleaning up temporary files...`

Plugin Change Detection (changedetection 1.0.1):
`Could not execute GRASS script /webodm/plugins/changedetection/changedetection.py from /webodm/app/media/tmp/tmp71_k158a_grass_engine: Starting GRASS GIS... Creating new GRASS GIS location <location>... Cleaning up temporary files... Executing <python /webodm/plugins/changedetection/changedetection.py reference_pc=/webodm/app/media/project/2/task/8d15a99a-63e4-48e4-a564-d3c9de27551b/assets/odm_georeferencing/odm_georeferenced_model.laz compare_pc=/webodm/app/media/project/2/task/8672b8e1-3523-4e01-87a8-1f93561c519e/assets/odm_georeferencing/odm_georeferenced_model.laz reference_dsm=/webodm/app/media/project/2/task/8d15a99a-63e4-48e4-a564-d3c9de27551b/assets/odm_dem/dsm.tif reference_dtm=/webodm/app/media/project/2/task/8d15a99a-63e4-48e4-a564-d3c9de27551b/assets/odm_dem/dtm.tif compare_dsm=/webodm/app/media/project/2/task/8672b8e1-3523-4e01-87a8-1f93561c519e/assets/odm_dem/dsm.tif compare_dtm=/webodm/app/media/project/2/task/8672b8e1-3523-4e01-87a8-1f93561c519e/assets/odm_dem/dtm.tif aligned_dsm=/webodm/app/media/plugins/changedetection/8d15a99a-63e4-48e4-a564-d3c9de27551b_8672b8e1-3523-4e01-87a8-1f93561c519e_dsm.tif aligned_dtm=/webodm/app/media/plugins/changedetection/8d15a99a-63e4-48e4-a564-d3c9de27551b_8672b8e1-3523-4e01-87a8-1f93561c519e_dtm.tif format=GeoJSON epsg=4326 display_type=heatmap resolution=0.2 min_area=40.0 min_height=5.0 can_align_and_rasterize=false> ... Traceback (most recent call last): File "/webodm/plugins/changedetection/changedetection.py", line 111, in <module> from compare import compare File "/webodm/plugins/changedetection/compare.py", line 4, in <module> import cv2 ModuleNotFoundError: No module named 'cv2' Execution of <python /webodm/plugins/changedetection/changedetection.py reference_pc=/webodm/app/media/project/2/task/8d15a99a-63e4-48e4-a564-d3c9de27551b/assets/odm_georeferencing/odm_georeferenced_model.laz compare_pc=/webodm/app/media/project/2/task/8672b8e1-3523-4e01-87a8-1f93561c519e/assets/odm_georeferencing/odm_georeferenced_model.laz reference_dsm=/webodm/app/media/project/2/task/8d15a99a-63e4-48e4-a564-d3c9de27551b/assets/odm_dem/dsm.tif reference_dtm=/webodm/app/media/project/2/task/8d15a99a-63e4-48e4-a564-d3c9de27551b/assets/odm_dem/dtm.tif compare_dsm=/webodm/app/media/project/2/task/8672b8e1-3523-4e01-87a8-1f93561c519e/assets/odm_dem/dsm.tif compare_dtm=/webodm/app/media/project/2/task/8672b8e1-3523-4e01-87a8-1f93561c519e/assets/odm_dem/dtm.tif aligned_dsm=/webodm/app/media/plugins/changedetection/8d15a99a-63e4-48e4-a564-d3c9de27551b_8672b8e1-3523-4e01-87a8-1f93561c519e_dsm.tif aligned_dtm=/webodm/app/media/plugins/changedetection/8d15a99a-63e4-48e4-a564-d3c9de27551b_8672b8e1-3523-4e01-87a8-1f93561c519e_dtm.tif format=GeoJSON epsg=4326 display_type=heatmap resolution=0.2 min_area=40.0 min_height=5.0 can_align_and_rasterize=false> finished. Cleaning up temporary files...
`
Thanks for your help.

pierotofy, 2020-09-29:
Hi @taue2512,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-10-02 04:20) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

taue2512, 2020-09-29:
I've updated the post using the template accordingly. To me its an issue with the GIT as some required dependencies are not included in a fresh install.

taue2512, 2020-09-29:
What I've noticed is that after finishing an UPDATE process of WebODM Manager 1.2, the following error is thrown after first restart:
`
[32mwebapp      |[0m INFO Installing requirements.txt for [plugins.changedetection.plugin]
[32mwebapp      |[0m Collecting geojson==2.4.1
[32mwebapp      |[0m   Using cached geojson-2.4.1-py2.py3-none-any.whl (15 kB)
[32mwebapp      |[0m ERROR: Could not find a version that satisfies the requirement opencv-python==4.1.0.25 (from -r requirements.txt (line 2)) (from versions: 3.4.8.29, 3.4.9.31, 3.4.9.33, 3.4.10.35, 3.4.10.37, 3.4.11.39, 3.4.11.41, 3.4.11.43, 4.1.2.30, 4.2.0.32, 4.2.0.34, 4.3.0.36, 4.3.0.38, 4.4.0.40, 4.4.0.42, 4.4.0.44)
[32mwebapp      |[0m ERROR: No matching distribution found for opencv-python==4.1.0.25 (from -r requirements.txt (line 2))
[32mwebapp      |[0m WARNING Failed to install requirements.txt  for [plugins.changedetection.plugin]
`

atobiese, 2020-09-30:
See issue #903.  It occurs because opencv-python==4.1.0.25 fails being installed in the elevationmap and changedetection plugin. I quickly tried updating the opencv version but did not fix  it. I'll look at it some more when I get the chance.

taue2512, 2020-10-01:
> See issue #903. It occurs because opencv-python==4.1.0.25 fails being installed in the elevationmap and changedetection plugin. I quickly tried updating the opencv version but did not fix it. I'll look at it some more when I get the chance.

Thanks. I was noticing that in python3, the `import cv2` command works while in python2 it fails in the docker image.

pierotofy, 2020-10-21:
Should be fixed with #927. Please update and re-open if the issue persists? :pray:

closed 2020-10-21 by pierotofy