WebODM/WebODM #919: #Feature request: Clipping the results (Ortho/DTM/DSM/Contours) to project area boundaries

wearevr3d, 2020-10-02:
This is to request a long needed and easy feature to limit the boundaries of a project to a specific area to create more presentable results.

Most users have to import WebODM's outputs such as Ortho/DTM/DSM/Contours into a third party software like QGIS/ArcGIS, add project boundary polygon, clipping rasters by a mask vector layer and cleaning up. 

Would be great to integrate this QGIS workflow in WebODM if any developer could please. Being able to import project boundaries as KML/CSV format to WebODM or defining project boundaries on the screen in Orthophoto and clipping all the results within that project boundary would add a lot more presentation value to the output files.

Here's the link to the topic discussion in the community as reference: https://community.opendronemap.org/t/is-it-possible-to-limit-the-boundaries-of-what-is-visualized-in-webodm-via-gps-coordinates/2486/4

pierotofy, 2020-10-02:
Hi @wearevr3d,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-10-05 03:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Richardsonmapper, 2020-10-10:
This would be a very helpful update indeed!

W6oody, 2021-04-07:
I agree, this would be a good addition. My QGIS experience  is limited and for me being able to "tweak" results in WebODM before exporting would be huge. Almost an additional step to the pipeline as a "Tidy up" stage. Just my thoughts.

Ruthalas, 2025-04-12:
Thank you so much for adding this feature!
One thing I wanted to note was that it appears the cropped boundaries are not applied when using the openaerialmap plugin to submit to OpenAerialMap.

I'm not sure if that is something to work with the plugin author on or not, but the plugins don't appear to have links to github repos, so I am not sure how to go about that.
@pierotofy (if I have guessed the username correctly) appears to be the author of that plugin though.

Regardless, thank you!

pierotofy, 2025-04-12:
Glad you like it! Yes it might be best to open a separate issue 🙏