WebODM/WebODM #921: Feature Request: View Current Location

Saijin-Naib, 2020-10-04:
I would like to see the ability for WebODM to take advantage of modern browser Geolocation APIs to allow for instant locating of the current user using a button on the GUI, with similar UX to established norms.

I believe this could be a candidate for a smaller funding drive.

Thread:
https://community.opendronemap.org/t/current-position-webodm/5548

pierotofy, 2020-10-04:
Hi @Saijin-Naib,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-10-07 16:15) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

chisquare88, 2020-10-05:
this function must need **https** protocol not http.

reference: [https://developer.mozilla.org/en-US/docs/Web/API/Geolocation_API](https://developer.mozilla.org/en-US/docs/Web/API/Geolocation_API)

pierotofy, 2020-10-05:
This is a cool idea for a plugin; we'd welcome pull requests! Anyone interested?

commanderwall, 2020-10-05:
Its a great idea! If it can work as a plugin.

commanderwall, 2022-02-01:
> This is a cool idea for a plugin; we'd welcome pull requests! Anyone interested?

Hello, any news about this? What can I do to get some energy to this "view current location" case? I am not competent enough for this, but can help and found if anyone would like to make a plugin for this function.

Saijin-Naib, 2022-02-01:
No movement yet, unfortunately.

You might know some people who could help?

closed 2024-08-01 by Saijin-Naib