WebODM/WebODM #923: Processing Nodes Offline

abrbhat, 2020-10-15:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 81 on Windows 10

### What is the problem?

Processing Nodes are being shown as offline even though they are available. Even refresh does not change the status.

[image]

Clicking on each node results in showing the correct status.
[image]

### What should be the expected behavior?

Nodes should always display correct status.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Not sure how to replicate.

pierotofy, 2020-10-15:
The nodes are refreshed every 30 seconds (or as you found out, if you visit the node's page).

So do the nodes come back online after 30 seconds? 

If not, you might have run out of memory and the worker process in charge of refreshing the node statuses has been likely terminated, preventing the refresh to happen.

abrbhat, 2020-10-17:
How can I observe refreshing of nodes? I think the worker process might have gone down. How to bring it back up?

pierotofy, 2020-10-17:
Press the "Retry" button when creating a task.

abrbhat, 2020-10-19:
Shows the same error and the node status is not refreshed.

pierotofy, 2020-10-19:
> If not, you might have run out of memory and the worker process in charge of refreshing the node statuses has been likely terminated, preventing the refresh to happen.

abrbhat, 2020-10-19:
Does not look like I'm low on memory. This is diagnostic for the WebODM instance. I'm running ClusterODM and NodeODM on separate instances.
[image]

pierotofy, 2020-10-21:
Try:

`./webodm.sh restart`

?

It's possible that the worker gets killed when you process a task.

abrbhat, 2020-10-23:
Tried this, seems good till now

closed 2020-10-25 by smathermather