WebODM/WebODM #925: Error in running tests

abrbhat, 2020-10-21:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox

### What is the problem?

1 test in the test suite is failing.

### What should be the expected behavior?

Tests should pass

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I have installed WebODM natively in an Ubuntu 18.04 virtual machine. I am trying to run tests through
```
python manage.py test
```

The test suite is running fine except one test. I've pasted the log here https://paste.ubuntu.com/p/vYWdsyH8RV/

pierotofy, 2020-10-21:
Do the tests pass if you launch them from the docker image?

It's likely GRASS is not installed properly.

pierotofy, 2020-12-16:
Closing due to inactivity.

closed 2020-12-16 by pierotofy