WebODM/WebODM #930: Feature Request - Move Tasks between Projects

Saijin-Naib, 2020-11-04:
I think it'd be great to be able to move a Task between Projects, likely from the Task's Edit page.

This can help with keeping things organized, as you may not realize you'll need/want new Projects immediately, and it will save you having to either to export/import, or re-upload data (if you've not processed it).

For instance, I'd now like to move my Pix4D example datasets out from the Github Project, which initially was just a generic "Test" Project.
[image]

pierotofy, 2020-11-04:
Hi @Saijin-Naib,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-11-07 17:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2021-10-14:
This has been implemented! :clinking_glasses:

closed 2021-10-14 by pierotofy