WebODM/WebODM #931: Cesium Ion plugin fails to upload

duckblaster, 2020-11-05:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

https://whatismybrowser.com/w/4QJYWQD

### What is the problem?

Attempting to upload a task to Cesium Ion using the plugin fails silently.
The Cesium Task UI shows it sitting at Uploading 0% for a few minutes, but then silently disappears.

### What should be the expected behavior?

Cesium Task UI shows upload progress and completes successfully.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Enable Cesium Ion plugin, configure token, click the Tile in Cesium Ion button on any task, select any output to upload.

Error logs below from `./webodm.sh start`

```
worker      | [2020-11-05 20:17:57,669: ERROR/MainProcess] Received unregistered task of type 'plugins.cesium-ion.api_views.upload_to_ion'.
worker      | The message has been ignored and discarded.
worker      |
worker      | Did you remember to import the module containing this task?
worker      | Or maybe you're using relative imports?
worker      |
worker      | Please see
worker      | http://docs.celeryq.org/en/latest/internals/protocol.html
worker      | for more information.
worker      |
worker      | The full contents of the message body was:
worker      | b'[["fae57a5d-8660-459c-9419-f0c2c4b39261", "ORTHOPHOTO", "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJqdGkiOiJlNjlmODI2ZC1kYTQ4LTQ3ZDktYjU3Yy02NjNlNWNkYjc2NGUiLCJpZCI6MzcwODcsImlhdCI6MTYwNDYwNTY5M30.l9-xqcdbSeVKj2YdTGIjT8cN3by-2_IrCfgplYfEBCc", "/webodm/app/media/project/2/task/fae57a5d-8660-459c-9419-f0c2c4b39261/assets/odm_orthophoto/odm_orthophoto.tif", "Te Kowhai | Task of 2020-11-05T03:24:42.623Z \\u2060\\u2014 Orthophoto", "", "", {"baseTerrainId": "", "textureFormat": false}], {}, {"callbacks": null, "errbacks": null, "chain": null, "chord": null}]' (554b)
worker      | Traceback (most recent call last):
worker      |   File "/usr/local/lib/python3.6/dist-packages/celery/worker/consumer/consumer.py", line 559, in on_task_received
worker      |     strategy = strategies[type_]
worker      | KeyError: 'plugins.cesium-ion.api_views.upload_to_ion'
```

pierotofy, 2020-11-05:
I think the Cesium plugin has been broken for a while now; unless somebody steps up and fixes it, it's likely we'll end up removing it in the near future.

mateo3d, 2021-03-16:
cross-post from ODM community:

I got the cesium-ion plugin working again. 2 caveats:

It doesn’t place 3d models properly (othos and DEMs are perfect). I’ve used my own datasets as well as samples from ODM github to no avail; 3d models always end up within 500m of where they should be just not exact. Adjust Location feature in Cesium after the upload is tedious but not difficult. I’m willing to try other datasets if someone has one that was known to work before the plugin broke.

If you use Cloud Import for a project, the Cesium plugin buttons do not appear at all. Other projects without Cloud Import still function properly.

Is it better to do a pull request now to share the code changes or get these bugs worked out first?

pierotofy, 2022-12-14:
Closing this, we're still waiting for a fix on the plugin and would welcome a PR. :pray:

closed 2022-12-14 by pierotofy