WebODM/WebODM #932: WebODM does not start on Fedora

nyalldawson, 2020-11-08:
I consistently see the same issue when trying to use webodm under Fedora. I've seen this across different workstations and different Fedora builds, so it's unlikely a local issue.

Running `webodm.sh start` gets stuck at a certain point in an endless loop of repetitions of:

```
worker      | psql: could not connect to server: No route to host
worker      | 	Is the server running on host "db" (172.18.0.4) and accepting
worker      | 	TCP/IP connections on port 5432?
webapp      | psql: could not connect to server: No route to host
```

The full log looks like this:
```
./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
Starting node-odm ... failed
Starting db       ... failed
Starting broker   ... failed
Starting worker   ... failed
Starting webapp   ... failed
ERROR: No containers to start
Creating network "webodm_default" with the default driver
Creating broker            ... done
Creating db                ... done
Creating webodm_node-odm_1 ... done
Creating worker            ... done
Creating webapp            ... done
Attaching to webodm_node-odm_1, broker, db, worker, webapp
broker      | 1:C 08 Nov 2020 23:13:58.176 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker      | 1:C 08 Nov 2020 23:13:58.176 # Redis version=6.0.9, bits=64, commit=00000000, modified=0, pid=1, just started
broker      | 1:C 08 Nov 2020 23:13:58.176 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
broker      | 1:M 08 Nov 2020 23:13:58.176 # You requested maxclients of 10000 requiring at least 10032 max file descriptors.
broker      | 1:M 08 Nov 2020 23:13:58.176 # Server can't set maximum open files to 10032 because of OS error: Operation not permitted.
broker      | 1:M 08 Nov 2020 23:13:58.176 # Current maximum open files is 1024. maxclients has been reduced to 992 to compensate for low ulimit. If you need higher maxclients increase 'ulimit -n'.
db          | LOG:  database system was shut down at 2020-11-08 22:45:53 UTC
broker      | 1:M 08 Nov 2020 23:13:58.187 * Running mode=standalone, port=6379.
broker      | 1:M 08 Nov 2020 23:13:58.187 # Server initialized
broker      | 1:M 08 Nov 2020 23:13:58.187 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
db          | LOG:  MultiXact member wraparound protections are now enabled
broker      | 1:M 08 Nov 2020 23:13:58.188 * Ready to accept connections
db          | LOG:  autovacuum launcher started
db          | LOG:  database system is ready to accept connections
webapp      | psql: could not connect to server: No route to host
webapp      | 	Is the server running on host "db" (172.19.0.3) and accepting
webapp      | 	TCP/IP connections on port 5432?
worker      | psql: could not connect to server: No route to host
worker      | 	Is the server running on host "db" (172.19.0.3) and accepting
worker      | 	TCP/IP connections on port 5432?
webapp      | Postgres is unavailable - sleeping
worker      | Postgres is unavailable - sleeping
webapp      | psql: could not connect to server: No route to host
[... 89 lines cut ...]
worker      | 	TCP/IP connections on port 5432?
worker      | Postgres is unavailable - sleeping
webapp      | Postgres is unavailable - sleeping
worker      | psql: could not connect to server: No route to host
worker      | 	Is the server running on host "db" (172.19.0.3) and accepting
worker      | 	TCP/IP connections on port 5432?
webapp      | psql: could not connect to server: No route to host
webapp      | 	Is the server running on host "db" (172.19.0.3) and accepting
webapp      | 	TCP/IP connections on port 5432?
worker      | Postgres is unavailable - sleeping
webapp      | Postgres is unavailable - sleeping
worker      | psql: could not connect to server: No route to host
worker      | 	Is the server running on host "db" (172.19.0.3) and accepting
worker      | 	TCP/IP connections on port 5432?
webapp      | psql: could not connect to server: No route to host
webapp      | 	Is the server running on host "db" (172.19.0.3) and accepting
webapp      | 	TCP/IP connections on port 5432?
worker      | Postgres is unavailable - sleeping
webapp      | Postgres is unavailable - sleeping
webapp      | psql: could not connect to server: No route to host
webapp      | 	Is the server running on host "db" (172.19.0.3) and accepting
webapp      | 	TCP/IP connections on port 5432?
.... etc

```

pierotofy, 2020-11-08:
Hi @nyalldawson,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-11-11 18:20) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

nyalldawson, 2020-11-08:
@pierotofy thanks -- i've added the requested details

pierotofy, 2020-11-09:
It's probably not a bug with WebODM per-se, but perhaps a problem with running docker on Fedora. 

Possibly related: https://community.opendronemap.org/t/installation-on-fedora-31/3302/4

nyalldawson, 2022-06-12:
This is no longer an issue for me (on Fedora 36) -- I'm guessing subsequent OS updates have resolved the underlying incompatibility!

closed 2022-06-12 by pierotofy