WebODM/WebODM #935: WebODM Lightning error 정리

oneseungyeon, 2020-11-13:
WebODM Lightning을 사용하면서 발생한 에러들과 그에 대한 해결방법을 정리해 올립니다~

[image]

pierotofy, 2020-11-13:
Hi @oneseungyeon,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-11-15 22:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2020-11-14:
Please re-open the issue with a better description of the error? :pray:

closed 2020-11-14 by pierotofy