WebODM/WebODM #938: Webodm freezes

grdaveygrd, 2020-11-25:
Browser and operating system - Chrome 86 on Windows 10

Problem - periodic freezes when using Webodm - a task freezes up and cannot be stopped or deleted. This is an issue I've seen reported before, but it keeps happening. I'm using drone photographs with approximately 80% overlap, usually this works fine, but every now and then I have the problem and have to try and stop a task after many hours. There is no notification given that the it has frozen - but times like 17 hours are obviously excessive. I have a Core I7 computer with 32 GB RAM. The last time this happened I was trying to use fast orthophoto I think it is called, but problems have happened with different methods.

What I usually do is uninstall Webodm and Git and reinstall - as after one of these freezes Webodm won't start in my browser. Now I can't reinstall Webodm because when I try to  Clone Existing Repository I get directed to a Github sign in. When I sign in, the process crashes.

pierotofy, 2020-11-25:
Hi @grdaveygrd,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-11-27 21:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

grdaveygrd, 2020-11-26:
Hello,

I'm trying to report bugs, so I think this is the correct place. One of the apparent bugs (in something) means that I can no longer install Webodm so cannot reproduce the original problems. It seems that this email from Github may behind the installation problems - _A first-party GitHub OAuth application (Git Credential Manager) with gist, repo, and workflow scopes was recently authorized to access your account_. When I try to install Webodm I get a Github sign in screen, as I try to sign in I get a Git Gui error - Clone failed. Cannot fetch branches and objects. See console output for details.

The original problem with Webodm is that approximately every 5 times I use it, it freezes. The last time this was with approximately 200 drone photographs with approximately 80% overlap. I use the same image overlap each time and either fast orthophoto or high resolution and can't see any apparent reason for the intermittent problems. Is my 32gb of RAM insufficient? These images are readily stitchable in MS ICE. When freezing happens - indicated by far too many hours of processing, and nothing happening on the progress bar, I do not seem to be able to either stop or delete the tasks. I cannot then start Webodm and the only way to proceed that I can see is to reinstall it - now seemingly not possible.

Thanks

pierotofy, 2020-11-27:
Hey @grdaveygrd  :hand: it's quite likely that you haven't allocated enough RAM to docker. https://docs.docker.com/docker-for-windows/#advanced

This is a common installation/setup problem, but most likely not a bug in WebODM. Let's move the conversation to https://community.opendronemap.org if you need further help? :pray:

closed 2020-11-27 by pierotofy