WebODM/WebODM #942: Incorrect Layer/Task Names Showing in WebODM 'Map View'

ludeykrus, 2020-12-14:
Howdy, 

I have noticed that the layer/task names are not correct in the map view for WebODM. I can create a project and run multiple tasks with different dates as the names for each task. If I click on the "View Map" button at the top right of the list of tasks in the Project list, it will load a map with the different tasks as layers that can be toggled on and off. However, it appears most all of the tasks/layers are the exact same name, usually the bottom/oldest task. This is causing some confusion with others. I believe this is a bug, as I would expect the actual assocated task names to be displayed. 

Newest WebODM release running on Windows 10 Pro, installed via Docker, browser is Chrome, I have a test account on a spare domain where you can log in and see an example of this issue, if you wish (you will notice all layers show up as 2019-05-25 despite the layers having different names):

http://sininsteel.com:8000
User: Testing
PW: WebODMTest

pierotofy, 2020-12-14:
Hi @ludeykrus,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-12-17 16:05) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2020-12-16:
Thanks for the report!

This should be fixed with https://github.com/OpenDroneMap/WebODM/pull/943/commits/f99ee751710a27df7689670d7e36ad59546a7573 which will be included in the next release.

pierotofy, 2020-12-23:
Please update, see if the problem is solved and reopen if the issue persists? :pray:  Thanks!

closed 2020-12-23 by pierotofy