WebODM/WebODM #951: DEM/Orthophoto - fine details missing when compared to Metashape

CarlosGrohmann, 2021-01-07:
### What's your browser and operating system? 

Firefox 82 on macOS (Mojave)

### What is the problem?

I'm running WebODM and Metashape on a landslide dataset to check for similarities/differences. I don't have GCPs for this dataset.

In Metashape, I used the "high" option for sparse and dense clouds. The DEM has 6cm/px resolution and the orthophoto, 3cm/px (automatically calculated)

In ODM, I used the "ultra" option for `pc-quality`, `feature-quality`, and set the DEM resolution to 5cm and 3cm for the ortophoto.

At a first glance, the DEM and the orthophoto produced by ODM seem OK, but when zooming in, I can see that the ones produced by Metashape have more fine detail. The shaded relief of the ODM DEM is quite noisy and there are visible seams:

#### Orthophotos
Metashape
[image]  

ODM
[image]  

Metashape
[image]  

ODM
[image]  

Metashape
[image]  

ODM
[image]  

#### Shaded reliefs
Metashape
[image]  

ODM
[image]  

Metashape
[image]  

ODM
[image]  

Metashape
[image]  

ODM
[image]  

### What should be the expected behavior?

I would love to replace Metashape with ODM in my research. Just need help tuning the parameters maybe?

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

The dataset can be accessed [here](https://www.dropbox.com/s/h44ztyp72nskgs0/landslide_cgrohmann.zip?dl=0). Just note that this is part of unpublished research, so please keep it to yourself and contact me if you would like to use it for any publication.

The processing report from Metashape is [here](https://www.dropbox.com/s/fjc290awru4u1bq/landslide_report.pdf?dl=0). 

ODM processin options:
`resize-to: -1, rerun-from: opensfm, min-num-features: 8000, feature-quality: ultra, matcher-neighbors: 8, depthmap-resolution: 1000, opensfm-depthmap-method: BRUTE_FORCE, skip-3dmodel: true, ignore-gsd: true, mesh-size: 600000, mesh-octree-depth: 12, pc-quality: ultra, pc-las: true, pc-filter: 0, texturing-outlier-removal-type: gauss_damping, texturing-tone-mapping: gamma, dsm: true, dem-resolution: 5, orthophoto-resolution: 3, gps-accuracy: 5, optimize-disk-space: true`

pierotofy, 2021-01-07:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2021-01-07 by pierotofy

OKaluza, 2022-06-28:
Just adding a link to the discussion for anyone finding this https://community.opendronemap.org/t/dem-orthophoto-fine-details-missing-when-compared-to-metashape/6401

smathermather, 2022-07-22:
Super useful. Thanks for adding that link.