WebODM/WebODM #952: Metashape-style report?

CarlosGrohmann, 2021-01-07:
Is there a Metashape-style report available from ODM/WebODM? With processing time, number of points in the point clouds, vertices in the meshes, etc? 

As an example, see [here](https://www.dropbox.com/s/fjc290awru4u1bq/landslide_report.pdf?dl=0).

Thanks

pierotofy, 2021-01-07:
Hi @CarlosGrohmann,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-01-08 10:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

CarlosGrohmann, 2021-01-07:
Moving to Community, can close IMO

closed 2021-01-07 by pierotofy