WebODM/WebODM #960: macOS Big Sur (intel hardware) postgres problem during install. Effects Docker install and paid installer.

and-viceversa, 2021-02-22:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Safari 14 on macOS (Big Sur) and/or
Firefox 85 on macOS (Big Sur)

Hardware is Intel chipset, not M1.

### What is the problem?

Unable to install WebODM using Docker or the paid macOS installer. Both install processes enter the same infinite loop due to a problem loading Postgres. Please see the screen grab from WebODM Manager 1.2. I get the same "Postgres is unavailable - sleeping" and "psql: fe_sendauth: no password supplied" messages through both the Docker install and the paid installer. I originally thought I had done something wrong through the GitHub instructions, however since the paid installer has the same result I'm posting the issue.

[image]

### What should be the expected behavior?

Docker install and/or paid installer complete successfully.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I'm using a brand new iMac Pro. Out of the box I installed macOS updates, Homebrew, and some other projects and dependencies. I'm new to Docker, but in my understanding each Docker container is self sufficient so it's impossible that I've polluted the Postgres environment etc. Not sure what other info I can provide to reproduce since this is during the install process.

Let me know what else I can provide. Thank you.

pierotofy, 2021-02-22:
Mm, it's a strange error. I'm running some tests on MacOS right now. But like you say, docker provides a containerized environment that should prevent exactly these kinds of errors (if it works on Linux/Windows, it should also work on MacOS).

Do you have a firewall installed/running?

What happens if you press the "Update" button?

and-viceversa, 2021-02-22:
Running "Update" from WebODM Manager 1.2 returns the following:

```
Waiting for WebODM to start...
Press Start to launch WebODM
Updating[5A[2K
Stopping webapp            ... [32mdone[0m
[5B
[2A[2K
Stopping webodm_node-odm_1 ... [32mdone[0m
[2B
[4A[2K
Stopping worker            ... [32mdone[0m
[4B
[1A[2K
Stopping db                ... [32mdone[0m
[1B
[3A[2K
Stopping broker            ... [32mdone[0m
[3B
Removing worker            ... 
Removing broker            ... 
Removing db                ...
[1A[2K
Removing db                ... [32mdone[0m
[1B
[2A[2K
Removing broker            ... [32mdone[0m
[2B[3A[2K
Removing worker            ... [32mdone[0m
[3BRemoving network webodm_default
error while removing network: network webodm_default id 48e24d2a66b93b84475782d53bae5f7c1ba929560978b2d5819e46ecd48acf2a has active endpoints
Pulling db       ... 
Pulling broker   ... 
Pulling worker   ... 
Pulling node-odm ... 
Pulling webapp   ...
[3A[2K
Pulling worker   ... pulling from opendronemap/webodm_...
[3B
[3A[2K
Pulling worker   ... digest: sha256:d36de474ac19b1e76f...
[3B[3A[2K
Pulling worker   ... status: image is up to date for o...
[3B[3A[2K
Pulling worker   ... [32mdone[0m
[3B
[5A[2K
Pulling db       ... pulling from opendronemap/webodm_db
[5B
[1A[2K
Pulling webapp   ... pulling from opendronemap/webodm_...
[1B[5A[2K
Pulling db       ... digest: sha256:437e8772fddeb7e740...
[5B[5A[2K
Pulling db       ... status: image is up to date for o...
[5B
[1A[2K
Pulling webapp   ... digest: sha256:d36de474ac19b1e76f...
[1B[1A[2K
Pulling webapp   ... status: image is up to date for o...
[1B
[5A[2K
Pulling db       ... [32mdone[0m
[5B
[1A[2K
Pulling webapp   ... [32mdone[0m
[1B
[2A[2K
Pulling node-odm ... pulling from opendronemap/nodeodm
[2B[2A[2K
Pulling node-odm ... digest: sha256:937472a6f28b917947...
[2B[2A[2K
Pulling node-odm ... status: image is up to date for o...
[2B
[2A[2K
Pulling node-odm ... [32mdone[0m
[2B
[4A[2K
Pulling broker   ... pulling from library/redis
[4B
[4A[2K
Pulling broker   ... digest: sha256:f29bcfb891678a0c6a...
[4B
[4A[2K
Pulling broker   ... status: image is up to date for r...
[4B
[4A[2K
Pulling broker   ... [32mdone[0m
[4B
WebODM has updated. Press Start to launch WebODM
 WebODM...
From git://github.com/OpenDroneMap/WebODM
 * branch            master     -> FETCH_HEAD
Already up to date.
Stopping webapp            ... 
Stopping worker            ... 
Stopping broker            ... 
Stopping webodm_node-odm_1 ... 
Stopping db
```

and-viceversa, 2021-02-22:
Reproduced from fresh Docker install.

```
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input --depth 1
cd WebODM
./webodm.sh start 
```

Returns...

```
labmac@labmac ~> cd programs/WebODM/
labmac@labmac ~/p/WebODM (master)> ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
Starting db       ... failed
Starting node-odm ... failed
Starting broker   ... failed
Starting worker   ... failed
Starting webapp   ... failed
ERROR: No containers to start
Pulling node-odm (opendronemap/nodeodm:)...
latest: Pulling from opendronemap/nodeodm
83ee3a23efb7: Pull complete
db98fc6f11f0: Pull complete
f611acd52c6c: Pull complete
f4ac42be59b8: Pull complete
aca652ad2d79: Pull complete
dbc7309406ed: Pull complete
8de4db8b30d7: Pull complete
e519ad24a346: Pull complete
994cd454fe27: Pull complete
c3589a4f16d6: Pull complete
e4c4671843af: Pull complete
509d94698d23: Pull complete
de97faefc879: Pull complete
Digest: sha256:937472a6f28b917947ce7459f59c911d6f1fc7e82dfdaf0b4aee65546eb4b17f
Status: Downloaded newer image for opendronemap/nodeodm:latest
Pulling db (opendronemap/webodm_db:)...
latest: Pulling from opendronemap/webodm_db
10a267c67f42: Pull complete
e9a920522e33: Pull complete
6888e696bd71: Pull complete
798096eed143: Pull complete
fb58419959b5: Pull complete
97f9ec09cb68: Pull complete
8a3c81c4f1ea: Pull complete
948206ff7803: Pull complete
1447edda3d55: Pull complete
7e356caad655: Pull complete
549029a9fb44: Pull complete
6ce96c0fe15b: Pull complete
[... 324 lines cut ...]
webapp      | Password for user postgres: 
webapp      | psql: fe_sendauth: no password supplied
webapp      | Postgres is unavailable - sleeping
worker      | Password for user postgres: 
worker      | psql: fe_sendauth: no password supplied
worker      | Postgres is unavailable - sleeping
webapp      | Password for user postgres: 
webapp      | psql: fe_sendauth: no password supplied
webapp      | Postgres is unavailable - sleeping
worker      | Password for user postgres: 
worker      | psql: fe_sendauth: no password supplied
worker      | Postgres is unavailable - sleeping
webapp      | Password for user postgres: 
webapp      | psql: fe_sendauth: no password supplied
webapp      | Postgres is unavailable - sleeping
worker      | Password for user postgres: 
worker      | psql: fe_sendauth: no password supplied
worker      | Postgres is unavailable - sleeping
^CGracefully stopping... (press Ctrl+C again to force)
Stopping webapp            ... done
Stopping worker            ... done
Stopping db                ... done
Stopping webodm_node-odm_1 ... done
Stopping broker            ... done
```

and-viceversa, 2021-02-22:
Stopping and restarting appears to succeed, but http://localhost:8000 cannot connect.

```
labmac@labmac ~/p/WebODM (master)> ./webodm.sh stop
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Stopping WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.nodemicmac.yml stop
labmac@labmac ~/p/WebODM (master)> ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
Starting db       ... done
Starting broker   ... done
Starting worker   ... done
Starting node-odm ... done
Starting webapp   ... done
labmac@labmac ~/p/WebODM (master)> 
```

[image]

pierotofy, 2021-02-23:
Could you try to remove the `Z` value from `docker-compose.yml`:

```
    volumes:
      - dbdata:/var/lib/postgresql/data:Z
```

Change it to:

```
    volumes:
      - dbdata:/var/lib/postgresql/data
```

Then issue `./webodm.sh restart` ?

I just realized that my MacOS machine (Mac Mini 2012) is not going to get the Big Sur update (ugh, thanks Apple), so I can't rule out a problem specific to the OS (but the fact that this happened on Fedora too suggests that it might be something else).

and-viceversa, 2021-02-23:
The change above with `./webodm.sh restart` returned the same issue with the postgres infinite loop. I'm running a `./webodm.sh rebuild` now to be sure.

and-viceversa, 2021-03-01:
I wouldn't consider this issue solved necessarily, but WebODM successfully installed/runs when I deleted my Docker install, and then ran the paid WebODM manager.

Here's what I tested overall:
- Both paid installer and `./webodm.sh start` fails with the above issue when Docker and `docker-compose` are installed from `brew`.
- Both paid installer and `./webodm.sh start` fails with the above issue when Docker Desktop is installed from Docker's homepage.
- Paid installer succeeds when Docker is not installed (so WebODM Manager installs Docker for you).

Let me know if there is any other feedback I can provide here. Thanks again!

pierotofy, 2021-03-01:
Thanks for the info! This helps narrow it down. I wonder if this is due to some recent update to docker (the installer uses an older version of docker).

pierotofy, 2021-05-11:
Closing, as this is probably some issue with the docker software, not WebODM per-se.

closed 2021-05-11 by pierotofy

beedawg1985, 2021-06-01:
I'm running: 
```
Model Name:	MacBook Pro
Chip:	Apple M1
macOS 11.1 (20C69)
```

Can confirm the below is still true:

```./webodm.sh start fails with the above issue when Docker and docker-compose are installed from brew.```
```./webodm.sh start fails with the above issue when Docker Desktop is installed from Docker's homepage.```

Getting the error:
```
...
worker      | Postgres is unavailable - sleeping
webapp      | Postgres is unavailable - sleeping
worker      | psql: error: could not translate host name "db" to address: Name or service not known
webapp      | psql: error: could not translate host name "db" to address: Name or service not known
worker      | Postgres is unavailable - sleeping
webapp      | Postgres is unavailable - sleeping
...
```

Is there a work around for this yet? ODM is basically pay-only for me right now...

pierotofy, 2021-06-01:
Apple M1 is still not supported yet.

and-viceversa, 2021-06-01:
For anyone arriving from search, ODM and NodeODM are not affected by this issue on Intel Macs. I was able to install them using their GitHub documentation. For some reason only WebODM is affected. As I noted above, you can succeed by deleting your Docker install and letting the paid WebODM manager install it for you.