WebODM/WebODM #964: Can't seem to upload images on EFS

TheDiscordian, 2021-02-25:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Browser: Brave 1.19.12 (Chrome 88 on Linux)
OS: Manjaro 20.2.1 (Server: Debian 10 Buster)

### What is the problem?

Using docker and running with `./webodm.sh start --media-dir /mnt/efs/pointclouds/` everything appears fine, WebApp seems happy, but when we upload something and the worker is supposed to engage, I get no output indicating anything happened until it gives up I guess and says:

Error:

```
worker      | INFO Removing Task [Wittengale Road - 11/28/2020] (fa2681e7-236f-44b5-a238-db0b66a22995)                                                            
worker      | [2021-02-25 20:03:00,416: INFO/ForkPoolWorker-1] worker.tasks.process_task[d96edd3f-299c-476b-be5f-87c130ed804d]: Removing Task [Wittengale Road - 11
/28/2020] (fa2681e7-236f-44b5-a238-db0b66a22995)                                                                                                                   
worker      | WARNING [Errno 2] No such file or directory: '/webodm/app/media/project/2/task/fa2681e7-236f-44b5-a238-db0b66a22995/'                               
worker      | [2021-02-25 20:03:00,426: WARNING/ForkPoolWorker-1] worker.tasks.process_task[d96edd3f-299c-476b-be5f-87c130ed804d]: [Errno 2] No such file or direct
ory: '/webodm/app/media/project/2/task/fa2681e7-236f-44b5-a238-db0b66a22995/'
webapp      | WARNING Not Found: /api/projects/2/tasks/fa2681e7-236f-44b5-a238-db0b66a22995/
```

I can confirm, I don't see any activity in `/mnt/efs/pointclouds/project`.

### What should be the expected behavior?

It should upload the images and the worker processes it as normal.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

1. Create an Amazon EC2 instance.
2. Mount an EFS volume. (fstab `redacted.efs.us-east-2.amazonaws.com:/ /mnt/efs/pointclouds nfs4 nfsvers=4.1,rsize=1048576,wsize=1048576,hard,timeo=600,retrans=2,noresvport,_netdev 0 0`)
3. Run the same command as I did, with the media store on EFS (`./webodm.sh start --media-dir /mnt/efs/pointclouds/`).
4. Try to upload a dataset.

TheDiscordian, 2021-03-02:
Not a WebODM issue, reverse proxy was rejecting anything above 1MB causing the issue. Thank you for your time, sorry for the invalid issue.

closed 2021-03-02 by TheDiscordian

pierotofy, 2021-03-03:
Thanks for sharing the solution! Glad you found the cause.