WebODM/WebODM #971: WebOdm hangs on compressing all.zip for some datasets/runs.

cwitting, 2021-04-05:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Chrome 89 on Linux

### What is the problem?

WebOdm hangs on compressing all.zip for some datasets/runs.
This issue seems very similar to what is reported here: https://community.opendronemap.org/t/processing-hangs-at-compressing-all-zip/2013/28
which should have been fixed by this pull request: https://github.com/OpenDroneMap/WebODM/pull/893
which references that gdal should be updated to 3.1. However on the newest master it seems that GDAL is no longer 3.1 but 3.0.
Running "dpkg -l | grep gdal" in the worker container gives "gdal-bin 3.0.4+dfsg-1~bionic0"
I also see a lot of these warnings:
WARNING Using legacy implementation (GDAL >= 3.1 not found)
when running "docker logs worker"

I am not that familiar with webodm, but as far as i can see this could be caused by this commit https://github.com/OpenDroneMap/WebODM/commit/ff989cc3203a79dfa62146789ed2842715cbd6ed#diff-dd2c0eb6ea5cfc6c4bd4eac30934e2d5746747af48fef6da689e85b752f39557
which changes the base from debian to ubuntu. (In the original issue it was mentioned that gdal 3.1 was depending on debian).

### What should be the expected behavior?

WebOdm succesfully completing.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I cannot reproduce this every time, it only happens on some particular runs of the same data set. retrying from scratch a couple of times eventually makes it finish.

pierotofy, 2021-04-05:
Does updating GDAL to 3.1 fix the issue?

cwitting, 2021-04-06:
Yes it appears to fix it, at least I did not see any hangs in some brief testing on the same datasets.
It seems that on ubuntu 18.04 the gdal version in the ppa:ubuntugis/ubuntugis-unstable repository is 3.0
However in the ubuntu 20.04 version the gdal version is 3.2.1.

I tried updating the base to ubuntu 20.04 which includes the newer version of gdal and did not result in a hang. (See attached patch here: https://gist.github.com/cwitting/cda92b226f7ccc63ffea9d7c4602d48a), however I also had to update some other packages as they did not exist in 20.04.

Let me know if you think this could be a permanent fix (I did not test if anything else changed) and I can open a PR.

pierotofy, 2021-04-06:
A pull request would be amazing @cwitting ! Thanks for looking into this.

PR #972 (2021-04-06): Use ubuntu 20.04 as base to ensure gdal version is >= 3.1
In ubuntu 18.04 the gdal version in ppa:ubuntugis/ubuntugis-unstable is only 3.0, but in 20.04 it is 3.2.1.

This should fix #971 and is related to #891 which needs gdal >= 3.1
closed 2021-04-06 by pierotofy