WebODM/WebODM #974: iframe ReferenceError: viewer is not defined

ChipwizBen, 2021-04-11:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 86 on Fedora Linux

### What is the problem?

Embedding WebODM in an iframe breaks the 3D viewer with: "ReferenceError: viewer is not defined"

Any example can be seen here (3D tab):
https://dashboard.aerosurvey.co.nz/public.cgi?S=491&T=237308d7-7ebc-4741-916b-5ff45e709903

While on ODM itself, it works fine:
https://survey.aerosurvey.co.nz/public/task/237308d7-7ebc-4741-916b-5ff45e709903/iframe/3d/

### What should be the expected behaviour?

WebODM should function correctly in an iframe. This was the behaviour before, but this broke somewhere between today's release and (I think) around October last year.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

See links.

pierotofy, 2021-04-30:
I just opened the link, but seems to work OK? https://dashboard.aerosurvey.co.nz/public.cgi?S=491&T=237308d7-7ebc-4741-916b-5ff45e709903

[image]

pierotofy, 2021-05-11:
Closing, cannot reproduce, no reply from author.

closed 2021-05-11 by pierotofy