WebODM/WebODM #975: Image tags in docker-compose

olivierdalang, 2021-04-12:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

WebODM's docker-compose don't specify image tags for services. This makes the stack fragile : 
- a stack created from the same docker-compose with the command may result in different versions if a new image was published in the mean time
- updating a repo and running docker-compose up will not result in a fully up to date stack
- if manually pulling the images (`webodm.sh update` does that by calling `docker-compose pull`), you cannot downgrade in case if get issues with the new version

### What should be the expected behavior?

In all docker-composes, all referenced images should specify a version tag (meaning that all those images should be versioned on dockerhub).

Note that this is quite easy to setup up with dockerhub's autobuild feature.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? 

N/A

pierotofy, 2021-04-12:
Hey @olivierdalang :hand: we actually do this on purpose to keep everyone on a "rolling release" (https://en.wikipedia.org/wiki/Rolling_release). It helps us identify bugs quickly and push the software forward instead of focusing on backward compatibility issues.

Users can always tag a release and build their own docker images if versioning is required.

pierotofy, 2021-04-12:
One thing it might be helpful to add however would be the capability to specify a different set of images to launch WebODM via `webodm.sh` / docker compose without editing the docker-compose.yml files. Currently that needs to be done by hand and a config file / flags would be helpful here.

olivierdalang, 2021-04-12:
@pierotofy Ok I get the point, but I find this a bit scary in production (esp. that it's not possible to revert an upgrade that would cause some issues).

What about using latest on master, and freeze the tags on releases ? This way, master is a rolling release (used by `webodm.sh update`), yet it's still possible to have deterministic versions if checking out tags.

pierotofy, 2021-04-12:
Sure! We'd welcome a PR that implements this.