WebODM/WebODM #977: WebODM freeze on tiles generation for big orthophotos

charnesp, 2021-04-16:
Hi everyone,

When we render big orthophotos (especially when rendering vegetal indices) the Gunicorn hangs for seconds.

I think it is because we do not specify the number of workers :

`gunicorn webodm.wsgi --bind unix:/tmp/gunicorn.sock --timeout 300000 --max-requests 250 --preload`

I would then recommend to change on `start.sh`, line 143 this Gunicorn initialization to:

`gunicorn webodm.wsgi --bind unix:/tmp/gunicorn.sock --workers $(grep -c '^processor' /proc/cpuinfo) --timeout 300000 --max-requests 250 --preload`

And also maybe to add en ENV VARIABLE for a customization of the `max-requests`.

What do you think?

Thank you!

pierotofy, 2021-04-16:
Hi @charnesp,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-04-17 03:40) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

charnesp, 2021-04-16:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Safari / Chrome / Linux / macOS 

### What is the problem?

When we render big orthophotos (especially when rendering vegetal indices) the Gunicorn hangs for seconds.

I think it is because we do not specify the number of workers :

gunicorn webodm.wsgi --bind unix:/tmp/gunicorn.sock --timeout 300000 --max-requests 250 --preload

### What should be the expected behavior?
No hang during tile rendering on the browser

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing?

I have a 5 Go orthophoto and render vegetation indices

I would then recommend to change on start.sh, line 143 this Gunicorn initialization to:

gunicorn webodm.wsgi --bind unix:/tmp/gunicorn.sock --workers $(grep -c '^processor' /proc/cpuinfo) --timeout 300000 --max-requests 250 --preload

And also maybe to add en ENV VARIABLE for a customization of the max-requests.

What do you think?

( Using this parameters : `gunicorn webodm.wsgi --bind unix:/tmp/gunicorn.sock --workers $(grep -c '^processor' /proc/cpuinfo) --timeout 300000 --max-requests 750 --preload` solved the issue )

Thank you!

pierotofy, 2021-04-16:
Hey @charnesp :hand: thanks for looking into this. gunicorn docs actually recommend to set this value to `2 x $(NUM_CORES)`, would you be interested in opening a pull request?

charnesp, 2021-04-16:
Hey @pierotofy 
No problem for the pull request.
Thank you for your work!

pierotofy, 2021-04-16:
Thank you for the contribution! :clinking_glasses:

closed 2021-04-16 by pierotofy