WebODM/WebODM #985: I deleted my account, I re-entered, unable to upload pictures, uninstalled the program that I paid for, I lost it, I downloaded the program back and installed it, unable to upload pictures, ask for help. Spread

chanthalungsy, 2021-05-13:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Type answer here]

### What is the problem?

[Type answer here]

### What should be the expected behavior?

[Type answer here]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Type answer here]

chanthalungsy, 2021-05-13:
Help me please

DroneguyAK, 2021-05-13:
The case was closed

pierotofy, 2021-05-13:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2021-05-13 by pierotofy

chanthalungsy, 2021-05-13:
[image]

I can't get back into the system because it says 5 photos can only be uploaded.
o create a map, press the "Select Images and GCP" button, or drag and drop some images into a project.

You need at least 5 images
Images must overlap by 65% or more
For great 3D, images must overlap by 83%
A GCP file is optional, but can increase georeferencing accuracy

440 files selected. Please check these additional options:

There are no usable processing nodes. Make sure that at least one processing node is reachable and that you have granted the current user sufficient permissions to view the processing node (by going to Administration -- Processing Nodes -- Select Node -- Object Permissions -- Add User/Group and check CAN VIEW PROCESSING NODE). If you are bringing a node back online, it will take about 30 seconds for WebODM to recognize it.

chanthalungsy, 2021-05-13:
Help me, I lost my money.
The email I logged in
lar_chan@yahoo.com

pierotofy, 2021-05-13:
Once again, could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1: