WebODM/WebODM #986: Broken ubuntugis repo

pvgmenegasso, 2021-05-13:
Hello, I cannot currently run ./webodm.sh rebuild because apparently the ubuntugis ppa cannot be found.
Any helps with this issue ? Are you able to reproduce it ?

pierotofy, 2021-05-13:
Hi @pvgmenegasso,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-05-14 15:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2021-05-14:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

pvgmenegasso, 2021-05-21:
Hey Piero, apparently it was a network environment issue. I was able to solve it. Thanks a lot and sorry for bothering :P

closed 2021-05-21 by pvgmenegasso

pierotofy, 2021-05-21:
Glad to hear! No worries, thanks for updating the issue thread.