WebODM/WebODM #987: Processing Node not available for non-Admin despite permissions

ntrip6, 2021-05-17:
Hello, I've been successfully using WebODM (via Docker on an Ubuntu VM) and one project processed beautifully using the Admin account. Now I've got a non-Admin user wanting to use it but they don't see any processing nodes. 
I gave them all permissions (and staff status, but not SU), and so they should be able to add/use nodes but they see none.
I added them using the existing node-1 settings that works for Admin but no luck.
I tried restarting the whole app (./webodm start) -- incidentally maybe restart should be an option from the SU Admin panel since it needs to be run pretty often.
As you can see in this screenshot I've now added many nodes but none show up on as Available. Thanks 
[image]

pierotofy, 2021-05-17:
Hi @ntrip6,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-05-18 16:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

smathermather, 2021-05-17:
Come on over to community.opendronemap.org for support questions. We try to keep the issue queue for bugs and enhancements.

closed 2021-05-18 by pierotofy

ntrip6, 2021-05-19:
Well I've tried a variety of things and this seems like perhaps a bug in the node permissions. If not maybe someone will explain in the community forums.

ntrip6, 2021-05-21:
Granting the user SU privileges solves it. Not ideal but it'll do for now.