WebODM/WebODM #988: Instructions for changing permissions for missing node

ntrip6, 2021-05-17:
WebODM (linux version on Docker, Ubuntu VM).
The instructions for changing the Permissions on Node are out of date. Currently says

`60` files selected. Please check these additional options:
> There are no usable processing nodes.Make sure that at least one processing node is reachable and that you have granted the current user sufficient permissions to view the processing node (by going to Administration -- Processing Nodes -- Select Node -- Object Permissions -- Add User/Group and check CAN VIEW PROCESSING NODE). If you are bringing a node back online, it will take about 30 seconds for WebODM to recognize it.

**It should read:**
Administration > Accounts > (select) User> Permissions > then User permissions window
[image]

pierotofy, 2021-05-17:
Hi @ntrip6,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-05-18 18:25) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

ntrip6, 2021-05-18:
Alternately it should instruct you to add node processing permissions to the Default Group to apply to a new user as this isn't on by default.

pierotofy, 2021-05-19:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue providing **all** required information.

closed 2021-05-19 by pierotofy