WebODM/WebODM #990: Plant Health GeoTIFF Export Error

BigWaterUAV, 2021-05-27:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Safari 14.1 on macOS (Catalina), Chrome 91 on macOS (Catalina)]

### What is the problem?

[Export GeoTIFF From Plant Health Tab Does Not Work]

### What should be the expected behavior?

[Export GeoTIFF From Plant Health should be Functioning]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Tried Exporting GeoTIFF from Plant Health Tab and it did not work. Tried Exporting GeoTIFF from Surface Model and Terrain Model and it is Functional. 
Export of GeoTIFF from Plant Health results in a spinning wheel and then the following text: {"readyState":4,"responseText":"<!DOCTYPE html>\n<html lang=\"en\">\n<head>\n \n \n \n <meta charset=\"UTF-8\">\n <meta http-equiv=\"X-UA-Compatible\" content=\"IE=edge\">\n <meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">\n <meta name=\"author\" content=\"WebODM Contributors\">\n <link rel=\"icon\" type=\"image/png\" href=\"\" sizes=\"32x32\">\n\n <link rel=\"stylesheet\" type=\"text/css\" href=\"/static/app/css/bootstrap.min.css\" />\n\n <link href=\"/static/app/css/font-awesome.min.css\" rel=\"stylesheet\" type=\"text/css\">\n <link href=\"/static/app/css/metisMenu.min.css\" rel=\"stylesheet\">\n <link href=\"/static/app/css/sb-admin-2.css\" rel=\"stylesheet\">\n \n\n <script src=\"/static/app/js/vendor/modernizr-2.8.3.min.js\"></script>\n <script src=\"/static/app/js/vendor/es6-shim.min.js\"></script>\n <script src=\"/static/app/js/vendor/jquery-1.11.2.min.js\"></script>\n <script src=\"/static/app/js/vendor/system.js\"></script>\n <script src=\"/jsi18n/\"></script>\n\n \n\n \n <script type=\"text/javascript\" src=\"/static/app/bundles/main-881c9bce980d82122aaf.js\" ></script>\n<link type=\"text/css\" href=\"/static/app/bundles/css/main-881c9bce980d82122aaf.css\" rel=\"stylesheet\" />\n\n \n <script src='/plugins/changedetection/main.js'></script>\n<script src='/plugins/measure/main.js'></script>\n<script src='/plugins/elevationmap/main.js'></script>\n<script src='/plugins/cloudimport/load_buttons.js'></script>\n<script src='/plugins/osm-quickedit/main.js'></script>\n<script src='/plugins/contours/main.js'></script>\n<script src='/plugins/openaerialmap/main.js'></script>\n<script src='/plugins/lightning/add_cost_estimate.js'></script>\n<script src='/plugins/fullscreen/main.js'></script>\n <link href='/plugins/cloudimport/build/ImportView.css' rel='stylesheet' type='text/css'>\n<link href='/plugins/cloudimport/build/TaskView.css' rel='stylesheet' type='text/css'>\n<link href='/plugins/openaerialmap/style.css' rel='stylesheet' type='text/css'>\n \n\n <title>Login - </title>\n\n <link rel=\"stylesheet\" href=\"/static/CACHE/css/76ef6fdf1a1c.css\" type=\"text/css\" />\n\n <style type=\"text/css\">\n \n </style>\n</head>\n<body data-admin-utc-offset=\"0\">\n\n \n <div id=\"wrapper\">\n \n\n <nav id=\"navbar-top\" class=\"navbar navbar-default navbar-static-top\" role=\"navigation\" style=\"margin-bottom: 0\">\n <div class=\"navbar-header\">\n <button type=\"button\" class=\"navbar-toggle\" data-toggle=\"collapse\" data-target=\".navbar-collapse\">\n <span class=\"sr-only\">Toggle navigation</span>\n <span class=\"icon-bar\"></span>\n <span class=\"icon-bar\"></span>\n <span class=\"icon-bar\"></span>\n </button>\n \n [image]\n <a class=\"navbar-link\" href=\"/\"><p class=\"navbar-text\"></p></a>\n </div>\n\n \n </nav>\n\n \n<div style=\"text-align: center;\">\n <h3>500 Internal Server Error</h3>\n <h5>Something happened. The server logs contain more information.</h5>\n [image]\n</div>\n\n\n\n </div>\n\n \n \n \n\n</body>\n<script src=\"/static/app/js/vendor/metisMenu.min.js\"></script>\n<script>\n$(function(){\n $('#side-menu').metisMenu();\n\n $(window).bind(\"load resize\", function() {\n var topOffset = 50;\n var width = (this.window.innerWidth > 0) ? this.window.innerWidth : this.screen.width;\n if (width < 768) {\n $('div.navbar-collapse').addClass('collapse');\n topOffset = 100; // 2-row-menu\n } else {\n $('div.navbar-collapse').removeClass('collapse');\n }\n\n var height = ((this.window.innerHeight > 0) ? this.window.innerHeight : this.screen.height) - 1;\n height = height - topOffset;\n if (height < 1) height = 1;\n if (height > topOffset) {\n $(\"#page-wrapper\").css(\"min-height\", (height) + \"px\");\n }\n });\n\n var url = window.location;\n var element = $('ul.nav a').filter(function() {\n return this.href == url;\n }).addClass('active').parent();\n\n while(true){\n if (element.is('li')){\n element = element.parent().addClass('in');\n } else {\n break;\n }\n }\n\n if (element.parent().is('li')) element.parent().addClass('active');\n});\n\n</script>\n<script src=\"/static/app/js/vendor/bootstrap.min.js\"></script>\n\n</html>","status":500,"statusText":"Internal Server Error"}
![Plant Health Export GeoTiff Response]
(https://user-images.githubusercontent.com/84913265/119831339-cc923980-bec2-11eb-9b86-423438eb0092.png)
]

rionlerm, 2021-05-31:
Chrome -> Version 91.0.4472.77 (Official Build) (64-bit) and both exports (ph and ortho) work for me. Windows 10 OS.

markFieldman, 2021-08-17:
Confirmed in my case

markFieldman, 2021-08-17:
[image]
[image]

pierotofy, 2022-02-28:
Is this still happening with the latest version?

BigWaterUAV, 2022-02-28:
Sorry, I have not tested the new version yet. 
I’ll update and get back to you.

Thanks

Troy

	
Troy R. Lutz
Owner | Big Water UAV Solutions LLC. 
P.O. Box 309 LaPointe,WI 54850
719-966-9963
bigwater-uav-solutions.com
***@***.***

waruthp, 2022-03-01:
VARI GeoTIFF Raw map export error after latest updates

closed 2022-03-01 by pierotofy

waruthp, 2022-03-02:
> It should be fixed, if it's not feel free to reopen? 🙏

It was fixed already. Thanks