WebODM/WebODM #991: GCPEditorPro-WebODM-Plugin on Native Windows WebODM

ecoinfo, 2021-06-02:
I successfully upload the GCPEditorPro-WebODM-Plugin.zip but it did not show on the page of Manage Plugins.

[image]

pierotofy, 2021-06-02:
Hi @ecoinfo,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-06-02 22:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2021-06-15:
Should be fixed. Update and test? Thanks for the report! :pray:

closed 2021-06-15 by pierotofy