WebODM/WebODM #992: 3D model is not displaying

dolnz, 2021-06-03:
Hi Team,
This issue is pertaining to the new windows native install.  I don't see the issue for this version listed anywhere yet.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 88.0.1 (64-bit)
Windows 10 (newest update)
ODM Native Windows install, version 1.9.0 (with an update release applied 03 June 21).

### What is the problem?

After processing the image set and displying the output, the 3D viewer won't show the model.  The viewer opens fine but no model displys.

The assets do create a number of files including texture and point cloud outputs with can be viewed in other applications such as Cloud Compare or Meshlab.  So the data is there but it's just not viewing.

Note the 2D view is fine, with the expected accuracy based on the GPS on the drone.

### What should be the expected behavior?

When arriving in the 3D viewer the point cloud model should display ready for inspection.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Screenshots of the settings are downloadable here:  http://gofile.me/49FxH/DupYVIdPG
The assets for these settings are downloadable here:  http://gofile.me/49FxH/OavxJCXnF
(Downloads are valid till 18/06/21)

Note the settings in the screenshots are adjusted for a second processing attempt from the default settings, which yielded the same result of a no show.

dolnz, 2021-06-03:
**ADDED 3D Screen Display**  (03 June 21 10.00pm)
http://gofile.me/49FxH/p16iTUxyG

trygvefossum, 2021-06-03:
This might be helpful. We had problems similar to this, the issue was potree.js, specifically the check for a spatial reference system, which did not exist for the dataset. We had to comment out the check for the point cloud to display properly. 
 
This specific line was causing problems: You can check yourself using the inspection tool in your web-browser. 
`
			if (info.srs.wkt) {
				if (!this.projection) this.projection = info.srs.wkt;
				else this.fallbackProjection = info.srs.wkt;
			}
`

dolnz, 2021-06-03:
Hi, Thanks for the update.  Regarding the inspection tool, bear in mind this is the native windows version which is not running in a browser.  However the code must still be there for potree and in the source code.

Can I expect a release with the repair sometime soon, to update the native windows version?

pierotofy, 2021-06-03:
Based on the task log, for some reason, the `pc_ept` option did not turn on.

```
[INFO]    pc_ept: False
```

Which is required to show the 3D model.

Usually this is set automatically:

```

![image](https://user-images.githubusercontent.com/1951843/120671385-5449ec00-c45f-11eb-8163-74fbf9e01386.png)

For some reason in your case this didn't happen.

Did you change any settings?

dolnz, 2021-06-04:
No, not for the first time I ran the job which failed to display.  After that I changed some settings, more for the quality of the PC.  These are recorded in the screenshots linked in the previous notes.

I've noticed the same problem also, in the micmac processing node in WEBODM (browser version).

On that note MicMac is not available in the Windows Native version of WEBODM.  Will that be implemented at some time?

Hope this helps.  Cheers.

dolnz, 2021-06-04:
I have just finished processing a new run on the same data set and it works fine.  Great result in the output.

It seems the default setting for the pc_ept must have been set to false.  I will know to look out for it in the future.  Also it is probably one of those checklist things that needs to be recorded for final confirmation before rollout of a release.

Thanks for your help team.  It is much appreciated.  👍
Is there anything you want me to check before you close the issue?

pierotofy, 2021-06-05:
I guess I'm still puzzled by the fact that `pc_ept` was not turned on. Like I mentioned, this should be done automatically by the software (users should not need to set it explicitly). 

If you find a way to reproduce this, could you let us know and re-open the issue? :pray:

closed 2021-06-05 by pierotofy

dolnz, 2021-06-13:
I have replicated the issue again tonight.  It appears, the pc_ept is false by default.  See images attached.  It is the same for the 3D Imagery and the Default setting.

[image]
[image]
[image]

dolnz, 2021-06-13:
This was on buld 14

pierotofy, 2021-06-13:
The task settings shouldn't matter (in theory); NodeODM will always set `pc_ept` if it's not set when sending a task from WebODM. https://github.com/OpenDroneMap/NodeODM/blob/master/libs/Task.js#L72

pierotofy, 2021-06-13:
Can you post a screenshot of your processing node information?