WebODM/WebODM #998: "Commandline too long" WebODM native WIN 1.9.1

mcmuller, 2021-06-10:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
https://whatismybrowser.com/w/LMGFXLB
WIN10 Pro 21H1
AMD Ryzen 9 3900X
64GB RAM
500GiB Diskspace

### What is the problem?
Dataset with 591pics stops with "commandline too long", already applied "Long Filename Registry Patch", see attached log
[webodm-process too long-console.txt](https://github.com/OpenDroneMap/WebODM/files/6631535/webodm-process.too.long-console.txt)

### What should be the expected behavior?
...get it done

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
see log please!

pierotofy, 2021-06-10:
Same as https://github.com/OpenDroneMap/ODM/issues/1296 (which is getting fixed today, will make its way in WebODM by next week).

mcmuller, 2021-06-10:
ups, didn't see that one.

closed 2021-06-10 by mcmuller