WebODM/WebODM #1000: WebODM Message to User: Cannot process dataset

johngithens, 2021-06-13:
My sincere thanks to all who have contributed to this project. Most of my recent projects using WebODM have been successfully processed. However the latest has yet to be successful. I seek to understand the current limits, so that I do not attempt projects which will not succeed at this time. I do not expect, nor do I need a rapid response. 

I have had success with Pix4Dmapper on my enhanced Lenovo P-70 laptop to process collections of PPK-georeferenced images from a lab-calibrated camera, of a large shoreline terrain. In most cases in Pix4Dmapper I have not needed to create sub-projects to be merged for additional processing.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome, Version 91.0.4472.101 (Official Build) (64-bit)

### What is the problem?

Project terminates before successful completion, after selecting High Resolution. Project completes successfully when using Default parameters.

### What should be the expected behavior?

Project terminates after successful completion with High Resolution parameters
[image]
.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

See the second attachment, a view of the result when using Default parameters.
[image]

Link to folder on Google Drive with the image collection (will expire in 30 days): https://drive.google.com/drive/folders/1Djx1ODfvKeSLkDuNJmX3JjPxPVyrhdB-?usp=sharing

Project Name: SLB West and East - High Resolution - 31Mar2021
Uploaded Dataset: Sunlight Beach Road - 3/31/2021
Dataset image count: 475  
Created on: 6/12/2021, 2:13:29 PM
Processing Node: node-odm-1 (auto)
Options: ignore-gsd: true, dsm: true, pc-quality: high, dem-resolution: 2.0, orthophoto-resolution: 2.0, rerun-from: odm_report

Clues are probably in the following from the Task Output view:

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying.html
Traceback (most recent call last):
File "C:\WebODM\resources\app\apps\ODM\stages\odm_app.py", line 83, in execute
self.first_stage.run()
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 338, in run
self.next_stage.run(outputs)
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 338, in run
self.next_stage.run(outputs)
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 338, in run
self.next_stage.run(outputs)
[Previous line repeated 6 more times]
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 319, in run
self.process(self.args, outputs)
File "C:\WebODM\resources\app\apps\ODM\stages\odm_dem.py", line 91, in process
commands.create_dem(
File "C:\WebODM\resources\app\apps\ODM\opendm\dem\commands.py", line 192, in create_dem
run('gdalbuildvrt "%s" "%s"' % (tiles_vrt_path, '" "'.join(map(lambda t: t['filename'], tiles))))
File "C:\WebODM\resources\app\apps\ODM\opendm\system.py", line 90, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

pierotofy, 2021-06-13:
This is a bug that should have been fixed with https://github.com/OpenDroneMap/ODM/pull/1297/commits/64c2d4322b56cd29328f520f100086aadfe943b2

Make sure you are running the latest version. Go to **Tools --> Check for Updates**.

johngithens, 2021-06-13:
Thanks! Attempted to update to 1.9.1 build 22.  I tried allowing both exe's for WebODM and WebODMManager in Windows Security > Firewall and network protection. See recurring message in screenshot. Log files attached.
[image]
[pgsql.log](https://github.com/OpenDroneMap/WebODM/files/6644686/pgsql.log)
[app.log](https://github.com/OpenDroneMap/WebODM/files/6644687/app.log)

pierotofy, 2021-06-14:
Mm, very strange. For some reason WebODM cannot authenticate with the database.

Try to completely uninstall WebODM, then make sure the C:\WebODM folder (or wherever WebODM is installed) is deleted. Then re-install the latest version of the setup?

This will delete the projects you previously processed. Save them if you need to by copying C:\WebODM\resources\app\data\media\projects somewhere.

johngithens, 2021-06-14:
Mystery solved. Upon the update prompt, I did not notice that the update install chose an external hard drive, not my C drive. WebODM is now processing as expected. Thanks again for your help.