WebODM/WebODM #1001: Processing error for preset 3D Model with Lightning node

DatSci01, 2021-06-14:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

https://whatismybrowser.com/w/2GXJEN2

### What is the problem?

When running a dataset of images (different datasets ranging from 56-60 images each) using the lightning node and the preset "3D Model", I receive the error [{"name":"mesh-octree-depth","error":"Invalid value 12 (out of range)"}] as soon as the process reaches the point to upload images to the node (after image resizing).

### What should be the expected behavior?

The process should run without error. It runs without error using the local node.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I tried three different datasets of images, with the same result. Each processes normally using the local node for processing, but I receive the error described above at exactly the same point (uploading images to node). The error occurs only when the lightning node is selected, it processes normally using the local node ("node-odm-1"). The "3D Model" option selects the following options: mesh-octree-depth: 12, use-3dmesh: true, pc-quality: high, mesh-size: 300000. From what I can see, the problem occurs because the preset "3D Model" selects the value 12 for the mesh-octree-depth. Looking at the description for mesh-octree-depth, a value of 12 is within limits for local node processing, but must be <= 11 when using Lightning processing. So, the problem appears to be the preset 3D Model isn't aware of the constraints of the processing node selected and tries to force the value 12.

DatSci01, 2021-06-14:
Since higher values for mesh-octree-depth should yield better results for 3D Models, limiting Lightning node processing to a value of 11 means that possibly better results can be achieved using a local node than the Lightning node. Can the constraint on Lightning node be changed to allow 12 to be used for mesh-octree-depth?

pierotofy, 2021-06-14:
This is explained in FAQ #4: https://webodm.net/faq

> Because we need to maintain a certain quality of service for all users, we limit a few resolution parameters such as the orthophoto and DSM resolution (1 cm/px maximum) on our cloud service.

If you need `12`, we recommend to run your own local node? :pray:

closed 2021-06-14 by pierotofy

DatSci01, 2021-06-14:
Piero,

Thanks for pointing out the faq. I wasn't aware that this was one of the
parameters being limited as addressed in that faq.

The bug (at least in my view) is that the user interface allows the user to
select incompatible options. In particular, it allows the user to select
Lightning processing and also select Options "3D Model" (which sets
"mesh-octree-depth" to 12, a value incompatible with the max value allowed
for Lightning node), but then either doesn't warn the user about the
incompatible options selected or doesn't automatically adjust the
"mesh-octree-depth" to the Lightning limit of 11. Instead, the user finds
out about the incompatible options a few minutes later when the processing
error occurs. Maybe instead of a bug this is more a matter of design
philosophy for the user interface.

Bill Cottingham

pierotofy, 2021-06-14:
Thanks Bill ! This is a very good point.

I've made some changes to the lightning nodes; now if the value exceeds `11`, it will be set to `11` without erroring out. :+1:

DatSci01, 2021-06-14:
Piero,

Thanks! WebODM is a great tool!

Bill Cottingham