WebODM/WebODM #1010: problem to load webodm data to cesium

kbitzi, 2021-07-04:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

I am using WebODM  1.9.2 Build 28

[Type answer here] 
Chrome 91 on Windows 10

### What is the problem?

[Type answer here]
Chrome 91 on Windows 10

### What should be the expected behavior?

[Type answer here]
You can not see the button "view ion ion"

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Type answer here]
you can see the following error's in the app.log

"python: [2021-07-04 08:19:01,131: INFO/MainProcess] Received task: app.plugins.worker.eval_async[d0e41add-dee9-455f-b571-5cc14692d5e7]  \r\n"
"python: [2021-07-04 08:19:01,135: ERROR/MainProcess] Task app.plugins.worker.eval_async[d0e41add-dee9-455f-b571-5cc14692d5e7] raised unexpected: KeyError(\"'__name__' not in globals\")\r\nTraceback (most recent call last):\r\n  File \"C:\\WebODM\\resources\\app\\apps\\grass78\\Python37\\lib\\site-packages\\celery\\app\\trace.py\", line 385, in trace_task\r\n    R = retval = fun(*args, **kwargs)\r\n  File \"C:\\WebODM\\resources\\app\\apps\\grass78\\Python37\\lib\\site-packages\\celery\\app\\trace.py\", line 650, in __protected_call__\r\n    return self.run(*args, **kwargs)\r\n  File \"C:\\WebODM\\resources\\app\\apps\\WebODM\\app\\plugins\\worker.py\", line 29, in eval_async\r\n    return ns[funcname](*args, **kwargs)\r\n  File \"file\", line 19, in upload_to_ion\r\nKeyError: \"'__name__' not in globals\"\r\n"

pierotofy, 2022-02-28:
The Cesium team might be working on a fix soon: https://community.cesium.com/t/cesium-ion-webodm-plugin/17030

pierotofy, 2022-07-07:
Closing in favor of #931 (duplicate)

closed 2022-07-07 by pierotofy