WebODM/WebODM #1011: Vulnerable JS Library

lucabonuccelli, 2021-07-05:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

Vulnerable JS Library  Medium (Medium) Vulnerable JS Library
Description The identified library bootstrap, version 3.3.1 is vulnerable.
URL
https://demo.webodm.org/static/app/js/vendor/bootstrap.min.js

Method GET
Evidence * Bootstrap v3.3.1

URL
https://demo.webodm.org/static/app/js/vendor/jquery-1.11.2.min.js

Method GET
Evidence jquery-1.11.2.min.js

Instances 2
Solution Please upgrade to the latest version of bootstrap.
CVE-2019-8331
CVE-2018-14041

CVE-2018-14040
CVE-2018-14042

### What should be the expected behavior?

No Vulnerability alert

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[ZAP Scanning Report-3.pdf](https://github.com/OpenDroneMap/WebODM/files/6765466/ZAP.Scanning.Report-3.pdf)

pierotofy, 2021-07-05:
Thanks @lucabonuccelli ! Would you be interested in contributing a PR to fix this?

Note that upgrading bootstrap might break some pages, so make sure to test the UI afterwards if you make the changes.

pierotofy, 2021-07-27:
Closing due to no interest.

closed 2021-07-27 by pierotofy