WebODM/WebODM #1012: Cannot login with admin on new install

evilGmonkey, 2021-07-06:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 91 on Windows 10 (ACTUALLY BROWSER IS BRAVE)

### What is the problem?

After clean installing from GIT the latest version, (5th July 2021), it is impossible to login as it refuses the admin user. I tried multiple times and it seemed like the user had not been created. 

### What should be the expected behavior?

should be able to login using admin and default password.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Grab the lastest version and install.

digitivup, 2021-07-28:
I think this is because you have not created any accounts. I had the same problem as you. You need to click on the drone icon on login interface and then you can initialize a new admin account.

pierotofy, 2021-10-14:
Closing, since there seems to be a solution and it's not a software bug. :pray: