WebODM/WebODM #1013: Cluster not being used

evilGmonkey, 2021-07-06:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Version 1.26.74 Chromium: 91.0.4472.124 (Official Build) (64-bit) on Windblows 10

### What is the problem?

Job does not properly use the cluster even though its the only node specified in processing nodes. 

### What should be the expected behavior?

I expect it to farm it out to the nodes. 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Deloy webodm,
deploy at least 2 nodes
deplot clusterodm and add the nodes
add the cluster address as teh processing node in webodm.

FYI a work around to make it work is to actually specify the address of teh cluster in sm-custer parameter. However a strange thing is doing that made it work for a day using just the IP address and port of teh cluster, but then today I get a strange error and I had to change that to include HTTP... which is very strange as it worked before without it and NOTHING else was changed

pierotofy, 2021-07-06:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

smathermather, 2021-07-07:
I actually asked him to open this as a bug report. If I understand correctly it's a regression if you have to specify the ClusterODM address in WebODM, as this should be auto adding. But perhaps I'm underthinking this.

pierotofy, 2021-07-07:
Ah! :)

So it's likely that this is not a regression because there haven't been many changes on that part of the logic (but maybe it is).

`sm-cluster` is set automatically (or should we say, ClusterODM tries to **guess** the proper value, which may or may not be correct, depending on the setup), if the `--split` option is set (was it the case here?). The guess is based on the the address that was used to reach ClusterODM. https://github.com/OpenDroneMap/ClusterODM/blob/090403ffbe3edac4e717980bbdbaa1cc45374e7f/libs/netutils.js#L26

I always recommend to set `--public-address` when launching ClusterODM, because sometimes networks get in the way. If not set, ClusterODM will assume that its IP/host is the one that was used to reach it. (If you've added `1.2.3.4` in WebODM as the IP to connect to ClusterODM, ClusterODM will use that). But add docker to the mix, and you'll see how this is not always 100% correct.

https://github.com/OpenDroneMap/ClusterODM/blob/090403ffbe3edac4e717980bbdbaa1cc45374e7f/libs/taskNew.js#L216

pierotofy, 2021-11-01:
Closing since there's been no follow-up and it's likely a config issue.

closed 2021-11-01 by pierotofy