WebODM/WebODM #1015: WebODM on macOS stop working.

PLSomebody, 2021-07-07:
macOS 11.4 (20F71), Safari 14.1.1 (16611.2.7.1.4)

Suddenly WebODM stops working.

When I'm using ./webodm.sh start 
I'm getting this:
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1

And nothing more. Trying to open localhost:8000 effects in message that browser can't connect to the host.
I was trying to reinstall Docker, WebODM - no change in behavior.

pierotofy, 2021-07-07:
Hi @PLSomebody,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-07-08 19:05) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

sushidub, 2021-07-15:
I'll see if I can provide what's needed here as I'm having almost the exact same issue as the original issuer.

### What's your browser and operating system?
Chrome 91 on macOS (Catalina)

### What is the problem?
As of a few weeks ago all of the following `start` variations stopped instantiating any builds of the respective docker images needed to run WebODM/NodeODM and subsequently won't run any containers -- all while failing ever so silently.

> `./webodm.sh start --ssl --ssl-cert "/Users/Indo/Sites/certs/webodm.pem" --ssl-key "/Users/Indo/Sites/certs/webodm-key.pem" --hostname webodm.dev`
`./webodm.sh start --ssl --hostname webodm.dev`
`./webodm.sh start`

NOTE: I can confirm all three of the above scripts with the exact same environment worked beautifully up until 2-3 weeks ago.

here's the actual play
```sh
 $ ./webodm.sh start --ssl --ssl-cert "/Users/Indo/Sites/certs/webodm.pem" --ssl-key "/Users/Indo/Sites/certs/webodm-key.pem" --hostname webodm.dev
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: webodm.dev
Port: 8000
Media directory: appmedia
SSL: YES
SSL key: /Users/Indo/Sites/certs/webodm-key.pem
SSL certificate: /Users/Indo/Sites/certs/webodm.pem
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

Will enable SSL (Manual)
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.ssl.yml -f docker-compose.ssl-manual.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.ssl.yml -f docker-compose.ssl-manual.yml up --scale node-odm=1

~/Sites/code-prj/OpenDroneMap/WebODM
 $ [indo] ! 
```

### What should be the expected behavior?
1. On initial `./webodm.sh start` I expect Web/NodeODM to initialize the respective docker workflows, either building out the images needed to run the containers needed for accessing WebODM through the browser, or starting up any containers previously stopped as a result of a `./webodm.sh down` command.
2. I don't expect process feedback to stop either (unless I run the start script with the detached flag). 
3. Expect that plugging the Host and Port values (from within the process feedback) into a browser should also open up the WebODM config site running within its docker container. 

### How can we reproduce this?
1. Get yourself an Intel based macbook pro running Catalina 10.15.7
2. `git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input --depth 1`
2. startup `Docker version 20.10.7, build f0df350` if its not already running
3. cd WebODM
4. `./webodm.sh start`

Curious if this has anything to do with the whole move to breaking `docker-compose` up into `docker compose`...

Screens
[image]

More local Docker info
```
$ docker info
Client:
 Context:    desktop-linux
 Debug Mode: false
 Plugins:
  buildx: Build with BuildKit (Docker Inc., v0.5.1-docker)
  compose: Docker Compose (Docker Inc., 2.0.0-beta.4)
  scan: Docker Scan (Docker Inc., v0.8.0)

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 3
 Server Version: 20.10.7
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 1
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runtime.v1.linux runc io.containerd.runc.v2
 Default Runtime: runc
 Init Binary: docker-init
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 5.10.25-linuxkit
 Operating System: Docker Desktop
 OSType: linux
 Architecture: x86_64
 CPUs: 4
 Total Memory: 10.72GiB
 Name: docker-desktop
 Docker Root Dir: /var/lib/docker
 Debug Mode: true
```

pierotofy, 2021-07-15:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue providing **all** required information.

sushidub, 2021-07-15:
Wow - I worked pretty hard on providing what you guys need to at least read thru. And I'm curious what part of the [ISSUE_TEMPLATE](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md) is missing on my part?

pierotofy, 2021-07-15:
Hey @sushidub :hand: sorry for closing the issue, your report was fine, but the IssueBot (a script using my username) is pretty inflexible and looks only at the original poster's message (which never got edited), so it decided to close the topic. I've reopened it.

sushidub, 2021-07-16:
I got this to work locally just as I would expect.

1. Took closer look at the `webodm.sh` startup file
2. Did some testing between `docker-compose` vs `docker compose` within the `start` function - thinking perhaps the latest (3.5.2 / v2.0.0-beta.6) could be causing issues. Both resulted in the same output - as it should. 
3. Noticed the  logical OR `||` separating a `docker-compose start` and `docker-compose up.` One would expect if the first fails (no existing containers to be found) the second would then be invoked. Hmmmm. 
  > In my non-expert experience with docker-compose I don't think I've ever used (perhaps 'bothered' is more accurate) with the `start` sub-command, as `up` is generally smart enough to pick up any and all slack that `start` would have accomplished.
4. Regardless, after removing the control operator and `start` everything worked beautifully.
5. Invoking `./webodm.sh down` and then re-running using the same change starts up all the expected containery

_NOTE: I did not test as -dev_

```
webodm.sh diff
268 - command="$command start || $command up"
268 + command="$command up"
```

Feel free to close and/or I can submit a PR with the change.

pierotofy, 2021-07-16:
Hey @sushidub a PR would be fantastic! I think you're right that `up` should take care of starting existing containers.

closed 2021-07-18 by pierotofy