WebODM/WebODM #1022: Initial camera view not always loading in firefox

OKaluza, 2021-07-15:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 89 on Ubuntu Linux

### What is the problem?

Regarding https://github.com/OpenDroneMap/WebODM/pull/999
Thanks for the new features! 
The only problem is when I have saved the initial camera view, it doesn't always load first time, if I refresh the 3d point cloud view several times it seems about 50/50 whether it loads the default view or the saved view - I tried in Chrome and it seems to load every time, so it must be a browser specific thing.

### What should be the expected behavior?

The saved view should load every time

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

1) Open WebODM in Firefox and load 3d view
2) Set the initial view to something obviously different from default
3) Go to Scene tab - hit [Set initial camera view] and save the view
4) Refresh the page several times, the view will often revert to the default

pierotofy, 2021-07-15:
Thanks for the report @OKaluza :pray: I will look into this.

pierotofy, 2021-07-15:
Ok this should be solved with https://github.com/OpenDroneMap/WebODM/commit/d3b664e883d670bec358a5f92bba581226dff61a

Please test to see if the problem is solved :pray: if not feel free to re-open this issue!

Thanks for the report.

closed 2021-07-15 by pierotofy

OKaluza, 2021-07-15:
Thanks @pierotofy works perfectly now!