WebODM/WebODM #1023: Windows Native ODM crashes when Aligning submodels

nick4451, 2021-07-17:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 89 on Windows 10

### What is the problem?

When setting split to a value > 0 the application fails when "Aligning submodels".

### What should be the expected behavior?

The application should align the submodels and process the dataset.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

1. Open native WIndows WebODM application
2. Add project
3. Select Images (I selected 16 in this small test project)
4. Select Default options
5. Click Edit and set split to some integer > 0 (I selected 7)
6. Set Resize Images to No
6. Click Review
7. Click Start Processing

pierotofy, 2022-02-28:
This should be fixed with the latest software updates in ODM. If the problem persists, feel free to re-open this :pray:

closed 2022-02-28 by pierotofy

nick4451, 2022-03-01:
Thank you for letting me know. I cannot tell if it works since the native
Windows version will no longer process the datasets. The Linux version
works fine. I guess I will have to write another defect.