WebODM/WebODM #1026: Improvement to save disk space

SCFred28, 2021-07-27:
I Piero, I notice after conducting a lot of tasks using WebODM latest version for Windows (1.9.2 Built 28), I saturate the disk space of my computer. It result WebODM cannot start or execute other tasks.

By investigating, I discover I can manually delete all the content of this folder to restart WebDOM: "C:\WebODM\resources\app\apps\NodeODM\data"

And I do not lose the result of all the tasks that were executed. But this is a hack, and I ignore if they're any collateral.

1- Is a clean way I am not aware is implemented to delete the content of this folder to save space;
2- If no clean way is existing, is it possible to consider the suggestion to have an official "data cleaning" functionality?

Thanks,

Fred

pierotofy, 2021-07-27:
Hi @SCFred28,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-07-28 12:15) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

SCFred28, 2021-07-27:
**What's your browser and operating system?**
Chrome 92 on Windows 10

**What is the problem?**

WebODM crash in the middle of a huge bunch of task processing. I notice it consume 100% of disk space.

**What should be the expected behavior?**

WebODM do not crash if missing disk space. And alternatively have a way to clean disk space.

**How can we reproduce this? (What steps did you do to trigger the problem?** 

Run a bunch of subsequent task analysis (42 in my case)

**Other information noticed on the side of this crash**

By investigating, I discover I can manually delete all the content of this folder to restart WebDOM: "C:\WebODM\resources\app\apps\NodeODM\data"
And I do not lose the result of all the tasks that were executed. But this is a hack, and I ignore if they're any collateral.

Cheers!
~ Your friendly FredBot

PS: I'm just an automated script that handle some boring task of the live of Fred, not a human being.

pierotofy, 2021-07-27:
Thanks for the feedback! You can actually use the `Reset Processing Queue` button from the `Tools` menu to free up the NodeODM queue, including disk space for tasks that have already processed.

A newer upcoming version (build 29) includes a more descriptive menu item for this function.

closed 2021-07-27 by pierotofy