WebODM/WebODM #1032: Dynamic 'Share' link genration

arpitpara, 2021-08-05:
Platform - WebODM windows 10 Pro Native

Issue:
I'm using the REST API services of 'WebODM' to create project and tasks. I'm getting a ' task_id ' in the process of this. 
[image]

URL = http://127.0.0.1:29800/public/task/task_id/map/
But this link won't work till I click on the 'Share' button to generate the link on 'View Map' feature on that particular task first.

I want either this link to be generated from the REST API itself or activated 'by default' from the code after completion of the task.

pierotofy, 2021-08-05:
Hi @arpitpara,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-08-06 08:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

arpitpara, 2021-08-05:
Browser info:
https://whatismybrowser.com/w/8PENN9T

closed 2021-08-07 by arpitpara