WebODM/WebODM #1033: Missing Google Maps tiles when orthophoto overlayed over Google Maps

tobymurray, 2021-08-08:
First of all, I'm brand new to both drones and producing maps and this software and the hardware accessible at consumer price points is astounding. Everyone I show a map to has their mind blown a little bit, so thanks team!

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 90 on Windows 10
Using WebODM 1.8.3 deployed via Docker Compose

### What is the problem?

On the "View Map" screen with the Orthophoto tab selected, some Google Maps tiles do not load leaving arbitrary seeming gray boxes.

### What should be the expected behavior?

All Google Map tiles that would be even partially visible should load under the orthophoto

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

It seems to be independent of the input parameters and does not occur for the other base layers. This happens for any maps I produce, regardless of physical location or dataset size. It _feels_ like it goes in diagonal stripes from bottom left to top right.

For me so far the reproduction steps are: view any orthophoto overlaid on Google Maps

[image]
[image]
[image]

pierotofy, 2021-09-09:
Mm, it's possible that this might be an issue (temporary?) with Google servers. There's not much we can do about that. Does it work with another basemap layer (e.g. openstreetma?)

pierotofy, 2021-11-01:
Closing assuming an issue with Google servers.

closed 2021-11-01 by pierotofy