WebODM/WebODM #1035: Feature Request: Set Child Process Priority from WebODM GUI

Saijin-Naib, 2021-08-11:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Windows Native

### What is the problem?
I think it would be great to have an option to change the child processing process priority from the WebODM GUI. On machines that have to "work" while processing, having the child processes set to something like Below Normal or Low can allow the workstation to remain responsive for the worker while processing continues (at a very slight processing speed penalty). Conversely, on a machine that doesn't need to remain responsive, one may want to crank it up to Realtime and eke out that last few percent.

### What should be the expected behavior?
Somewhere in the GUI (Tools menu? Processing Parameter setting (extend it to Linux/macOS [option for "nice" etc?]) the user will be able to set the processing priority for all the child processes that get spawned from WebODM.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Use current WebODM to process locally. Manually change the priority of things like the various python processes that spawn to feel the change in usability of the machine hosting nodeODM.

pierotofy, 2023-08-21:
I'm going to close this, as it's outside of the scope of WebODM to tweak system internals. :pray:

Saijin-Naib, 2023-08-21:
In the Linux world, isn't it common for tasks to set their nice value? This is analogous, I'd think.