WebODM/WebODM #1053: ODM community unreachable

APOS80, 2021-09-06:
### What's your browser and operating system? 

Chrome windows 

### What is the problem?

Community on opendronemap.org unreachable.

### What should be the expected behavior?

Getting in an able to read posts

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Go there

pierotofy, 2021-09-06:
Seems online now? https://community.opendronemap.org/

APOS80, 2021-09-06:
[image]

APOS80, 2021-09-06:
I’m in Sweden, might be an internet issue.

pierotofy, 2021-09-06:
Mm, got a confirmation from Italy as well. It seems to work from the US.

pierotofy, 2021-09-06:
Pinging @smathermather @indiajohnson-cm

APOS80, 2021-09-06:
I have no problem with the main page, just the community forum.

smathermather, 2021-09-06:
Looking into it. I have confirmed it can't be accessed outside the US.

closed 2021-09-08 by pierotofy