WebODM/WebODM #1056: WebODM process not generating requsted orthophoto-kmz

serbinsh, 2021-09-08:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Apple Mac OSX 10.14.6 (18G9323) [Mojave]
Chrome Version 92.0.4515.159 (Official Build) (x86_64)

### What is the problem?

I am requesting the creation of an orthophoto KMZ file as part of the output asset set, but only this specific asset is not created.  However the requested PNG is created.  This behavior is consistent between all my different projects. 

### What should be the expected behavior?

The processing workflow should generate a KMZ file of the resulting RGB orthophoto. At least according to my reading of the documentation

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Run a standard processing workflow, such as "High Resolution" but then edit the workflow to include options like PNG and KMZ.  

For example, see screenshot 

[image]

Perhaps I am missing something?  Or perhaps because of the higher requested resolution the generation doesn't occur?

```
10-4-27-58:WebODM sserbin$ git log -1
commit f86c4a03d2318f2dbb351fc5710ab9fde9f73488 (HEAD -> master, origin/master, origin/HEAD)
Merge: a061a58 5ce33b6
Author: Piero Toffanin <pt@masseranolabs.com>
Date:   Tue Sep 7 00:57:41 2021 -0400

    Merge pull request #1054 from yudeqang/master

    add celery result backend config: retry policy
```

Any feedback is appreciated. Thanks!

pierotofy, 2021-09-08:
Thanks @serbinsh for the report :pray: this should be fixed with #1057.

closed 2021-09-08 by pierotofy

serbinsh, 2021-09-08:
Thanks!  I will merge and try it out!

serbinsh, 2021-09-08:
Hmm @pierotofy where should I expect to find the KMZ?  I just re-ran a workflow with the latest code and didn't have any luck generating the KMZ 
[image]

[image]

[image]

```
10-4-27-58:WebODM sserbin$ git log -1
commit 3ce7cb3cef1496fb7b09f276303c0bff3f65bad7 (HEAD -> master, origin/master, origin/HEAD)
Merge: f86c4a0 7b33535
Author: Piero Toffanin <pt@masseranolabs.com>
Date:   Wed Sep 8 10:28:36 2021 -0400

    Merge pull request #1057 from pierotofy/kmz

    Show orthophoto KMZ in task assets
  ```

pierotofy, 2021-09-08:
Did you issue:

`./webodm.sh update && ./webodm.sh restart`?

serbinsh, 2021-09-08:
I definitely did ./webodm.sh update but I didn't do it as a single update/restart.  Let me try that!

serbinsh, 2021-09-08:
Hhmmmm... not really sure what I might be doing wrong but even after doing 

```
./webodm.sh update && ./webodm.sh restart
```

and yet still not getting the KMZ output :/

pierotofy, 2021-09-08:
Can you share the full task output of the task?

serbinsh, 2021-09-08:
Here it is. looks like there was a command issued to create it.  I am mapping the data to my host machine so was looking in the output assets folder but am a little confused as to why the timestamps didn't update for the same project. this is why I am going to try again with a new project 

[task_output.txt](https://github.com/OpenDroneMap/WebODM/files/7131739/task_output.txt)

pierotofy, 2021-09-08:
Ah, whoops! I've forgot to update NodeODM also. The KMZ is getting created, but is not included in the `all.zip` archive.

I've opened a fix in NodeODM.

serbinsh, 2021-09-08:
ahh! no worries and thanks for opening a fix!

serbinsh, 2021-09-09:
OK updated nodeODM, giving it another go

```
wolfmanodesktop:nodeodm sserbin$ git log -1
commit 3ce7cb3cef1496fb7b09f276303c0bff3f65bad7 (HEAD -> master, origin/master, origin/HEAD)
Merge: f86c4a0 7b33535
Author: Piero Toffanin <pt@masseranolabs.com>
Date:   Wed Sep 8 10:28:36 2021 -0400

    Merge pull request #1057 from pierotofy/kmz

    Show orthophoto KMZ in task assets
  ```

PR #1057 (2021-09-08): Show orthophoto KMZ in task assets
Closes #1056﻿