WebODM/WebODM #1058: ElevationMap extension not working

Majstor, 2021-09-08:
I use the of WebODM in podman containers. In normal way all run well, but when i enable ElevationMap, WebODM becomes very slow in reaction. It tries to download constantly some packages and compile them, but it fails. See the container's log :
```
WARNING: Discarding https://files.pythonhosted.org/packages/38/a9/cd39fd25df434b5d9451dc266c12b72f68282a2b9bd5d7b4aa2d57d6c20e/opencv-python-4.4.0.44.tar.gz#sha256=e87d88a820050c0e886c9add48eac2f80ff29207a98cca25869a6868c519daa4 (from https://pypi.org/simple/opencv-python/) (requires-python:>=3.6). Command errored out with exit status 1: /usr/bin/python /tmp/pip-standalone-pip-2c9imzu8/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-vmtui60f/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip 'numpy==1.11.3; python_version=='"'"'3.5'"'"'' 'numpy==1.13.3; python_version=='"'"'3.6'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"'' 'numpy==1.17.3; python_version>='"'"'3.8'"'"'' Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement opencv-python==4.4.0.44 (from versions: 3.4.10.37, 3.4.11.39, 3.4.11.41, 3.4.11.43, 3.4.11.45, 3.4.13.47, 3.4.14.51, 3.4.14.53, 3.4.15.55, 4.3.0.38, 4.4.0.40, 4.4.0.42, 4.4.0.44, 4.4.0.46, 4.5.1.48, 4.5.2.52, 4.5.2.54, 4.5.3.56)
ERROR: No matching distribution found for opencv-python==4.4.0.44
WARNING Failed to install requirements.txt for [coreplugins.elevationmap.plugin]
INFO Registered [coreplugins.elevationmap.plugin]
INFO Registered [coreplugins.fullscreen.plugin]
INFO Registered [coreplugins.measure.plugin]
INFO Registered [coreplugins.posm-gcpi.plugin]
INFO Booting WebODM 1.9.6
WARNING Debug mode is ON (for development this is OK)
INFO Registered [coreplugins.contours.plugin]
INFO Registered [coreplugins.diagnostic.plugin]
INFO Installing requirements.txt for [coreplugins.elevationmap.plugin]
Collecting geojson==2.4.1
  Using cached geojson-2.4.1-py2.py3-none-any.whl (15 kB)
Collecting opencv-python==4.4.0.44
  Using cached opencv-python-4.4.0.44.tar.gz (88.9 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python /tmp/pip-standalone-pip-2lzz81ak/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-0jxdz3hm/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip 'numpy==1.11.3; python_version=='"'"'3.5'"'"'' 'numpy==1.13.3; python_version=='"'"'3.6'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"'' 'numpy==1.17.3; python_version>='"'"'3.8'"'"''
       cwd: None
  Complete output (622 lines):
  Ignoring numpy: markers 'python_version == "3.5"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.6"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.7"' don't match your environment
  Collecting setuptools
    Using cached setuptools-58.0.3-py3-none-any.whl (816 kB)
  Collecting wheel
    Using cached wheel-0.37.0-py2.py3-none-any.whl (35 kB)
  Collecting scikit-build
    Using cached scikit_build-0.12.0-py2.py3-none-any.whl (73 kB)
  Collecting cmake
    Using cached cmake-3.21.2-py2.py3-none-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (20.3 MB)
  Collecting pip
    Using cached pip-21.2.4-py3-none-any.whl (1.6 MB)
  Collecting numpy==1.17.3
    Using cached numpy-1.17.3.zip (6.4 MB)
  Collecting packaging
    Using cached packaging-21.0-py3-none-any.whl (40 kB)
  Collecting distro
    Using cached distro-1.6.0-py2.py3-none-any.whl (19 kB)
  Collecting pyparsing>=2.0.2
    Using cached pyparsing-2.4.7-py2.py3-none-any.whl (67 kB)
  Building wheels for collected packages: numpy
    Building wheel for numpy (setup.py): started
    Building wheel for numpy (setup.py): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-0c25cex9/numpy_6eb01c0c59104fd3968e4888652c017f/setup.py'"'"'; __file__='"'"'/tmp/pip-install-0c25cex9/numpy_6eb01c0c59104fd3968e4888652c017f/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-w_jf4vrz
         cwd: /tmp/pip-install-0c25cex9/numpy_6eb01c0c59104fd3968e4888652c017f/
    Complete output (278 lines):
    Running from numpy source directory.
    blas_opt_info:
    blas_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib']
      NOT AVAILABLE
  
    blis_info:
    customize UnixCCompiler
      libraries blis not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib']
      NOT AVAILABLE
  
    openblas_info:
    customize UnixCCompiler
    customize UnixCCompiler
      libraries openblas not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib']
      NOT AVAILABLE
[... 694 lines cut ...]
INFO Booting WebODM 1.9.6
WARNING Debug mode is ON (for development this is OK)
INFO Registered [coreplugins.contours.plugin]
INFO Registered [coreplugins.diagnostic.plugin]
INFO Registered [coreplugins.fullscreen.plugin]
INFO Registered [coreplugins.measure.plugin]
INFO Registered [coreplugins.openaerialmap.plugin]
INFO Registered [coreplugins.posm-gcpi.plugin]
INFO Booting WebODM 1.9.6
WARNING Debug mode is ON (for development this is OK)
INFO Registered [coreplugins.contours.plugin]
INFO Registered [coreplugins.diagnostic.plugin]
INFO Registered [coreplugins.fullscreen.plugin]
INFO Registered [coreplugins.measure.plugin]
INFO Registered [coreplugins.openaerialmap.plugin]
INFO Registered [coreplugins.posm-gcpi.plugin]
INFO Booting WebODM 1.9.6
WARNING Debug mode is ON (for development this is OK)
INFO Registered [coreplugins.contours.plugin]
INFO Registered [coreplugins.diagnostic.plugin]
INFO Registered [coreplugins.fullscreen.plugin]
INFO Registered [coreplugins.measure.plugin]
INFO Registered [coreplugins.posm-gcpi.plugin]
```
Disable ElevatioMap and all come back to normal.

pierotofy, 2021-09-08:
Hi @Majstor,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-09-09 16:25) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

gromain, 2021-12-28:
I witness the exact same issue on a setup running vanilla WebODM started in docker with `./webodm.sh start` on a virtual KVM machine.

taue2512, 2022-02-01:
Same issue for me on the latest WebODM Manager / Docker image running on a MacOS machine. It worked for a longer time flawlessly but ceased to work sinice approximately 6 months. It applies for the installation for both elevationmap and changedetection plugins.

PR #1140 (2022-02-07): Fix plugins
Fixes #1058, #1059 without installing additional deps.﻿
closed 2022-02-07 by pierotofy