WebODM/WebODM #1059: Changedetection plugin not working

Egor-Morozov, 2021-09-10:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 92 on Ubuntu 18.04

### What is the problem?

Problem on running Changedetection plugin (big slowdown on activate it and error message in using it)

### What should be the expected behavior?

I use  WebODM 1.9.6 installed at ubuntu 18.04 by Docker. WebODM & Ubuntu are fresh & updated.
When i try to enable  Changedetection plugin -  WebODM becomes very slow in reaction. It tries to download constantly some packages and compile them, but it fails. See the log.txt

[log.txt](https://github.com/OpenDroneMap/WebODM/files/7143118/log.txt)

When i try to use  Changedetection plugin - i get error message. See the log2.txt
[log2.txt](https://github.com/OpenDroneMap/WebODM/files/7143176/log2.txt)

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

You can reproduce it at WebODM 1.9.6 installed at ubuntu 18.04 by Docker if you will try enable  Changedetection plugin
and if you will push button "Preview" in "View Map" view of any multitasks dataset.
The same type of  error i see in issue 1058 https://github.com/OpenDroneMap/WebODM/issues/1058

rumenmitrev, 2021-12-06:
I was not able to fix issue from above suggestions but , it looks like missing libraries and compile tools in the `worker` container.
Plugins are not able to build `opencv` , `scipi` , `numpy` .
Not quite sure that all tools and libraries below are nedded
I had mange it , with the following commands:
```bash
sudo /usr/bin/docker exec -it worker apt update
sudo /usr/bin/docker exec -it worker apt install -y make libblas-dev automake gcc g++ subversion python3-dev liblapack-dev gfortran libopenblas-dev libatlas-base-dev libblas3 liblapack3
sudo /usr/bin/docker exec -it worker pip install -r coreplugins/elevationmap/requirements.txt
sudo /usr/bin/docker exec -it worker sed -Ei 's/^scipy==(.*)/scipy>=\1/g' coreplugins/changedetection/requirements.txt
sudo /usr/bin/docker exec -it worker pip install -r coreplugins/changedetection/requirements.txt
```
Need to toggle  `changedetection` and `elevationmap` enable/disable

gromain, 2021-12-28:
@rumenmitrev I had to run those commands for the webapp container (running them only for the worker did not help). Otherwise it's great and this solved the issue!
Thanks for those.

Also solved #1058 for me.

Saijin-Naib, 2022-01-27:
So we need to adjust how our docker image is being constructed to include these missing libraries, right? That's the ideal solution from what I'm gathering here...

gromain, 2022-01-27:
Exactly!
Or maybe find a way to include the dependencies with the plugin package itself? I'm not sure how the plugin system works, so that's why I'm saying that, but there could be a better way from doing this than just blindly including big libraries in the image.

Saijin-Naib, 2022-01-27:
Thanks for confirming. This is above my head, but I'll try to take a look at this to see if maybe we can just lean on pop and requirements.txt files or something to pull the proper files.

gromain, 2022-02-01:
I'm testing some changes on the Dockerfile here. I'll need to do the same for NodeODM.

If you want to try:
```
FROM ubuntu:21.04
MAINTAINER Piero Toffanin <pt@masseranolabs.com>

ARG TEST_BUILD
ENV PYTHONUNBUFFERED 1
ENV PYTHONPATH $PYTHONPATH:/webodm
ENV PROJ_LIB=/usr/share/proj

# Prepare directory
ADD . /webodm/
WORKDIR /webodm

# Install Node.js
RUN apt-get -qq update && apt-get -qq install -y --no-install-recommends wget curl && \
    wget --no-check-certificate https://deb.nodesource.com/setup_12.x -O /tmp/node.sh && bash /tmp/node.sh && \
    apt-get -qq update && apt-get -qq install -y nodejs && \
    # Install Python3, GDAL, PDAL, nginx, letsencrypt, psql
    apt-get -qq update && apt-get -qq install -y --no-install-recommends python3 python3-pip python3-setuptools python3-wheel git g++ python3-dev python2.7-dev libpq-dev binutils libproj-dev gdal-bin pdal libgdal-dev python3-gdal nginx certbot grass-core gettext-base cron postgresql-client-13 gettext tzdata && \
    update-alternatives --install /usr/bin/python python /usr/bin/python2.7 1 && update-alternatives --install /usr/bin/python python /usr/bin/python3.9 2 && \
    # Install dependencies required to build elevationmap and changedetection plugins
    apt-get -qq install -y --no-install-recommends make libblas-dev automake gcc subversion liblapack-dev gfortran libopenblas-dev libatlas-base-dev libblas3 liblapack3 && \
    # Install pip reqs
    pip install -U pip && pip install -r requirements.txt "boto3==1.14.14" && \
    # Install requirements for elevationmap and changedetection
    bash -c "export MAKEFLAGS="-j$(nproc)" && pip install -r coreplugins/elevationmap/requirements.txt" && \
    sed -Ei 's/^scipy==(.*)/scipy>=\1/g' coreplugins/changedetection/requirements.txt && \
    bash -c "export MAKEFLAGS="-j$(nproc)" && pip install -r coreplugins/changedetection/requirements.txt" && \
    # Setup cron
    ln -s /webodm/nginx/crontab /var/spool/cron/crontabs/root && chmod 0644 /webodm/nginx/crontab && service cron start && chmod +x /webodm/nginx/letsencrypt-autogen.sh && \
    /webodm/nodeodm/setup.sh && /webodm/nodeodm/cleanup.sh && cd /webodm && \
    npm install --quiet -g webpack@4.16.5 && npm install --quiet -g webpack-cli@4.2.0 && npm install --quiet && webpack --mode production && \
    echo "UTC" > /etc/timezone && \
    python manage.py collectstatic --noinput && \
    python manage.py rebuildplugins && \
    python manage.py translate build --safe && \
    # Cleanup
    apt-get remove -y g++ python3-dev libpq-dev make libblas-dev automake gcc subversion liblapack-dev gfortran libopenblas-dev libatlas-base-dev && apt-get autoremove -y && \
    apt-get clean && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/* && \
    rm /webodm/webodm/secret_key.py

VOLUME /webodm/app/media
```

rumenmitrev, 2022-02-02:
FYI:
https://github.com/OpenDroneMap/WebODM/pull/1139

pierotofy, 2022-02-17:
Should be fixed :clinking_glasses: thanks for reporting the issue!

closed 2022-02-17 by pierotofy