WebODM/WebODM #1061: WebODM Flatpack/Snap?

APOS80, 2021-09-26:
### What's your browser and operating system? chrome Windows Ubuntu

### What is the problem?

Hard to get going on Linux

### What should be the expected behavior?

WebODM as Flatpack or Snap would make things easier.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Docker is for sysadmins

pierotofy, 2021-09-26:
Getting this to work on Snap/Flatpack is a significant effort; would you be interested in contributing toward this?

APOS80, 2021-09-27:
> Getting this to work on Snap/Flatpack is a significant effort; would you be interested in contributing toward this?

I don’t know where to begin. I now some python and racket. Very little C.

bluecamel, 2021-10-05:
I've just posted a [PR that adds a Debian package](https://github.com/OpenDroneMap/WebODM/pull/1066).  It's not exactly what you were asking for, but it might help if you are on a Debian-based system.

pierotofy, 2021-11-12:
Closing, feel free to open a PR that implements flatpack and/or snap.

closed 2021-11-12 by pierotofy