WebODM/WebODM #1064: Cannot update: certificate has expired

SCFred28, 2021-10-01:
### What's your browser and operating system?

Chrome 94 on Windows 10

### What is the problem?

Cannot update: certificate has expired

[image]

### What should be the expected behavior?

Continue to receive update.

### How can we reproduce this? (

With WebODM 1.9.7. build 32 click on the "Tools" Menu then "Check for Updates"

pierotofy, 2021-10-01:
Mm, strange! If you open https://webodm.org/test/ do you get a certificate error?

[image]

pierotofy, 2021-10-01:
Latest version is build `32` though, so good news is you're on the latest version.

SCFred28, 2021-10-01:
I give a try for a fresh reinstall.

------ Message d'origine ------

(#1064)

SCFred28, 2021-10-01:
Not working either. Do I need to renew the license? Is their any 
expiring?

------ Message d'origine ------
De:
À: "OpenDroneMap/WebODM" 
***@***.***>; 
"OpenDroneMap/WebODM" ***@***.***>
Cc : "Author" ***@***.***>
Envoyé : 2021-10-01 12:58:07
Objet : Re[2]: [OpenDroneMap/WebODM] Cannot update: certificate has 
expired (#1064)

***@***.***>

pierotofy, 2021-10-01:
No, no need to renew; it's a problem with our SSL certificate. We need to find a fix for it.

pierotofy, 2021-10-01:
https://www.zdnet.com/article/fortinet-shopify-others-report-issues-after-root-ca-certificate-from-lets-encrypt-expires/

SCFred28, 2021-10-01:
Apparently you have to switch to the latest available ISRG Root X1 valid until 2035-06-04 (intermediate R3)https://letsencrypt.org/certificates/Télécharger Outlook pour AndroidDe : Piero Toffanin ***@***.***>Envoyé : vendredi 1 octobre 2021 16 h 01À : OpenDroneMap/WebODMCc : SCFred28; AuthorObjet : Re: [OpenDroneMap/WebODM] Cannot update: certificate has expired (#1064)
https://www.zdnet.com/article/fortinet-shopify-others-report-issues-after-root-ca-certificate-from-lets-encrypt-expires/

SCFred28, 2021-10-02:
Help thread about this problem:https://community.letsencrypt.org/t/help-thread-for-dst-root-ca-x3-expiration-september-2021/149190Télécharger Outlook pour AndroidDe : Frédéric Bouchard ***@***.***>Envoyé : vendredi 1 octobre 2021 16 h 31À : OpenDroneMap/WebODM; OpenDroneMap/WebODMCc : AuthorObjet : Re: [OpenDroneMap/WebODM] Cannot update: certificate has expired (#1064)Apparently you have to switch to the latest available ISRG Root X1 valid until 2035-06-04 (intermediate R3)https://letsencrypt.org/certificates/Télécharger Outlook pour AndroidDe : Piero Toffanin ***@***.***>Envoyé : vendredi 1 octobre 2021 16 h 01À : OpenDroneMap/WebODMCc : SCFred28; AuthorObjet : Re: [OpenDroneMap/WebODM] Cannot update: certificate has expired (#1064)
https://www.zdnet.com/article/fortinet-shopify-others-report-issues-after-root-ca-certificate-from-lets-encrypt-expires/

jakevis, 2021-10-05:
+1 - have a repo here as well if you need anything