WebODM/WebODM #1069: When duplicating tasks, output point count is not updated when duplicated task is re-run

TimeScience, 2021-10-09:
I'm testing parameters for reconstructing point clouds from GoPro 360 camera images (by duplicating the same task multiple times) and I'm getting a bug where ODM reports the number of Reconstructed Points as 11,591, no matter what the settings and despite reporting large differences in the GSD and area surveyed. @smathermather [suggested](https://community.opendronemap.org/t/inaccurate-unchanging-point-counts-for-multiple-runs-of-a-project/9070) this was an issue with duplicated tasks not updating their stats correctly, rather than an issue with the specific images I was running, so I'm posting it here as a bug.

Link to images for testing: https://cloudstor.aarnet.edu.au/plus/s/cRvaQCGsswJnfmj 

[Link to image of issue](https://drive.google.com/file/d/1xkWfpxhGeXBkw2BmevYTRLyJPsLNXfFB/view?usp=sharing)

pierotofy, 2021-10-09:
Hi @TimeScience,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2021-10-09 21:35) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

closed 2021-10-09 by pierotofy

pierotofy, 2021-10-09:
Thanks @TimeScience :hand:, nice find, this should be fixed with the latest ODM update.

TimeScience, 2021-10-11:
Thanks @pierotofy